OpenHantek/OpenHantek6022 #96: Reoccurance of previous issue #113 from OpenHantek ( firmware upload failed)

daveychinook, 2020-06-09:
Hello, I am a newby to your code, looks very interesting and like you've had some good success.  
Sorry I am also new to GitHub and am not aware of the protocol to ask you for help.

Being brand new to your software I am not sure what is going wrong.  I've searched for my error and it seems this was something as described in the Original OpenHantek as issue [#113](https://github.com/OpenHantek/openhantek/issues/113), that being fixed long ago, I am not sure what I have done wrong.

I am running Debian (stable 9.12 stretch w/ Gnome) and sudo apt installed the pre-compiled 3.1.0 version from the .deb file openhantek_20200508-4209460_amd64.deb I downloaded from here on Github.

When I connect my DSO and run OpenHantek I get what looks to be normal behaviour loading firmware, in terminal I see:
$ OpenHantek
stop CPU
... WROTE: 792 bytes, 10 segments, avg 79
reset CPU
2nd stage: write external memory
stop CPU
2nd stage: write on-chip memory
... WROTE: 5092 bytes, 68 segments, avg 74
reset CPU

But then I get a popup GUI window that has a button to [Connect to First Device] with a upload failed message
[image]

On click of that button I get the failure message:
[image]
  Once I click [OK] the application ends and my command prompt returns.

My DSO6022BE hardware works normally under Win10 with the packaged software, but I want to get it working under Debian with your software, please Help.

 
**Computer environment (please complete the following information):**
 - OpenHantek version: v3.1.0 as pre-compiled in .deb file
 - OS: debian
 - Distribution, version: 9.12 stretch

**Scope device (please complete the following information):**
 - Device Hantek 6022BE

Ho-Ro, 2020-06-09:
Hi @daveychinook 
The log messages as well as the GUI window do not come from OpenHantek6022. The debug output of `stop CPU` and other similar messages is disabled and the *Connect to first device* window does not exist in the source code. Do you have also an old version (openhantek) installed?
OpenHantek6022 (that lives in this repo) and openhantek (living in https://github.com/OpenHantek/openhantek and being the ancestor) are different projects. What does `OpenHantek -v` say? Also the output of `which OpenHantek` could be helpful. 
If there is an old install it would be sufficient to delete the *wrong* binary `OpenHantek`, otherwise reinstall the deb from here, best with `sudo apt install --reinstall openhantek...deb` this pulls also all dependencies. If you use `sudo dpkg ...` you have to add a `sudo apt -f install` to fix the missing dependencies.

Good luck

> Sorry I am also new to GitHub and am not aware of the protocol to ask you for help.

You used exact the correct way to ask for help.

daveychinook, 2020-06-09:
Martin, 100% nail on head, I had originally tried to install the openhantek project a few months back, once I found your newer project I had shifted gears.  For some reason apt, apt --reinstall, dpkg  etc.. were not removing the old binary.  ( Clearly I am no package managing wiz :) )

Once I tracked the binary down ( type openhantek ) and removed that old binary and re-installed using apt with your .deb I am now seeing your tool, it looks like I am up and running.

Thank you so much for quickly responding and helping find that annoying issue.

closed 2020-06-09 by daveychinook