OpenHantek/OpenHantek6022 #97: Frequency Measurment ?

Courty40, 2020-06-21:
**Describe the bug**
Feeding the scope with a clean 10MHZ signal and its displays nicely but the Frequency field bottom right just shows 1.4khz. had a look through the doco and cant see where this is mentioned.

Im sure im being daft but cant see why this would read differently ?

**To Reproduce**
**IMPORTANT:** Always unplug and replug the scope if used before with a different application, e.g. sigrok or the original Hantek software.

Steps to reproduce the behavior:
1. Go to '...'
2. Select '....'
3. See error

**Expected behavior**
10mhz in and 10mhz displayed on the scope screen 

**Screenshots**
If applicable, add screenshots or a video to help explain your problem.

**Computer environment (please complete the following information):**
 - OpenHantek version: 3.10
 - OS: linuz
 - Distribution, version: debian stable amd64 Buster
 - Video hardware: Integrated Graphics
 - OpenGL version: New build, latest
 - Qt version: New build, latest
 - C++ compiler version: n/a - used RPM

**Scope device (please complete the following information):**
 - Device Hantek 6022BE
 - Program top line [e.g. *OpenHantek6022 (Build 3.1.0) - Device (FW0206)
 - Input signal: 10mhz, sine, 3.5vpp
 - Probe setting: probe X1

**Additional context**
Add any other context about the problem here, e.g. the program output if started in a console like xterm.

Ho-Ro, 2020-06-21:
Timebase?
Samplerate?
Please provide a screenshot that documents your settings

> had a look through the doco and cant see where this is mentioned

??

Anyeos, 2020-06-26:
There are on the bottom more than one frequency gauge. Make sure you are looking at the correct one. One is for the selection and other is for all the channel. The last have the "real" frequency as I guess you are expecting.

Ho-Ro, 2020-06-26:
> clean 10MHZ signal and its displays nicely

With the scope's max sample rate of 30 MS/s you get 3 (three!) samples per period - how can it display *nicely*? Please provide more info if you want help!

Ho-Ro, 2020-06-28:
@Courty40: No answer for a week gives me the impression that you are not interested in solving your issue - so I will close here.

closed 2020-06-28 by Ho-Ro