OpenHantek/OpenHantek6022 #99: Real time option

spam-receiver, 2020-06-23:
In real-time mode (when the trigger is disabled with a time base of 200 ms - 10 s), clear the screen from the previous graph every time before starting to draw a new graph. A button or a checkbox for quick on/off this feature.

Ho-Ro, 2020-06-26:
Please check the latest 3.1.1-rc3 ...

spam-receiver, 2020-06-27:
I really liked the scroll, but sometimes it’s easier to observe a static graph, so the previous option (without scroll) should also be available.

Ho-Ro, 2020-06-28:
I don't like the wipe mode, because it throws away valuable information: e.g. you measure with 1 s/div (yielding 10 seconds on the screen) for 12 seconds, in *roll mode* you'll see the last 10 seconds, in *wipe mode* you'll see the last two seconds only and 8 useless seconds with an empty trace.
The most intuitive solution for displaying slow and time-unlimited processes is the appearance of a [chart recorder](https://en.wikipedia.org/wiki/Chart_recorder) - and if you want a defined starting point I've added the *refresh* button that clears the screen and starts the trace again from the left margin. see 4ffe158, available as the last [unstable](https://github.com/OpenHantek/OpenHantek6022/releases/tag/unstable).

spam-receiver, 2020-06-28:
Well done! But what prevents the implementation of those two options? They do not interfere with other options, but only expand the possibilities of viewing waveforms in real time.

spam-receiver, 2020-06-28:
The Refresh button should only be active in Scroll mode when the time base is 200 ms or higher.

Ho-Ro, 2020-06-28:
> what prevents the implementation of those two options?

- Simple answer: I do not like it.
- More technical answer: The roll mode is already a dirty hack and each new mode adds more ugly code - and I have neither the time nor the desire to change these program structures from scratch.

(Have I already told that OpenHantek is based on code that is up to ten years old?)

spam-receiver, 2020-06-28:
Sorry, I didn’t know that everything is so complicated. Thanks for all, that you do!

Ho-Ro, 2020-06-28:
A question out of curiosity, what do you do with the program?

spam-receiver, 2020-06-28:
Use it with Hantek DSO-6022BL in development of some bluetooth devices.

Ho-Ro, 2020-06-30:
done with 3.1.1

closed 2020-06-30 by Ho-Ro