OpenHantek/OpenHantek6022 #103: Communication failure windows 10

michaelrempel, 2020-07-04:
**Describe the bug**
Red blinky light when software opens, Upload Firmware: Operation not supported or unimplemented on the platform. Did Hantek brick your upload?

**To Reproduce**
**IMPORTANT:** Always unplug and replug the scope if used before with a different application, e.g. sigrok or the original Hantek software.
Yup, that was done. I can load the Hantek software, and it sorta works as much as that software does.
I tried the Hantek driver as downloaded. That didnt want to work at all. I then signed the driver myself and it allowed the Hantek software to work.

The driver I installed was the one packaged with the program.

Steps to reproduce the behavior:
1. Go to '...'
2. Select '....'
3. See error

**Expected behavior**
A clear and concise description of what you expected to happen.
Green blinky light and working software able to connect.

**Screenshots**
If applicable, add screenshots or a video to help explain your problem.

**Computer environment (please complete the following information):**
 - OpenHantek version: [use always the latest commit, e.g. *release v2.03, commit 1a2b3c4, ...*]
 - OS: [e.g. *linux, macOS, windows, ...*]
 - Distribution, version: [e.g. *debian stable amd64, ...*]
 - Video hardware: [e.g. *Intel Xeon ... Integrated Graphics*]
 - OpenGL version: [e.g. *OpenGL 4.5, Mesa 19.0.3*]
 - Qt version: [e.g. *Qt 5.11.x*]
 - C++ compiler version: [e.g. *gcc 8.3.x*]

**Scope device (please complete the following information):**
 - Device  Hantek 6022BE
 - Program top line OpenHantek6022 b345 Winx64 
 - Input signal: [any/none
 - Probe setting: Ch1 probe X10, ch2 none

Ho-Ro, 2020-07-05:
It is very difficult for to follow your description and get the necessary info to help with your problem. Please be concise - all provided files have a name.

> I tried the Hantek driver as downloaded.

Which *"Hantek driver"*? Does it have a name? I do not provide a driver, the Windows port of my project *OpenHantek6022* links the executable *OpenHantek.exe* with the dynamic library *libusb-1.0.dll* to communicate with the hardware. For Windows you have to enable the USB access as [mentioned](https://github.com/OpenHantek/OpenHantek6022#run-openhantek6022) in the [documentation](https://github.com/OpenHantek/OpenHantek6022/blob/master/docs/build.md#windows). Did you follow these steps?

> That didnt want to work at all. 

What happens? 

> I then signed the driver myself and it allowed the Hantek software to work.
The driver I installed was the one packaged with the program.

What does *"the Hantek software"* mean - OpenHantek.exe? But then again, which *"driver packaged with the program"*?

Martin

michaelrempel, 2020-07-06:
I first tried the WinUsb driver, failed, then installed all the Zadig drivers, failed.
[image]
I tried just after install, fail. I tried after reboot, fail. The error message was the same both times with the WinUsb driver first, and then again with everything in Zadig.

michaelrempel, 2020-07-06:
My Zadig.log is quite long. I tried several times. Note I cleared it the first time, this is just what I did today.
Zadig 2.5.730
Windows 10 64-bit (Build 18363)
ini file 'zadig.ini' not found in 'S:\Electronics\Osciloscope' - default parameters will be used
default driver set to 'WinUSB'
0 devices found.
Using inf name: libusb-win32_Generic_Device.inf
Successfully extracted driver files.
Installing driver. Please wait...
libwdi:info [wdi_prepare_driver] K driver available - adding the libusbK DLLs to the libusb-win32 inf
libwdi:info [extract_binaries] successfully extracted driver files to C:\Users\MichaelR1\usb_driver
libwdi:info [wdi_prepare_driver] successfully created 'C:\Users\MichaelR1\usb_driver\libusb-win32_Generic_Device.inf'
libwdi:info [wdi_prepare_driver] Creating and self-signing a .cat file...
libwdi:debug [AddFileHash] 'libusb0.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\MichaelR1\usb_driver\amd64\libusb0.dll'
libwdi:debug [AddFileHash] 'libusb0.sys': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\MichaelR1\usb_driver\amd64\libusb0.sys'
libwdi:debug [AddFileHash] 'libusb0_x86.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\MichaelR1\usb_driver\amd64\libusb0_x86.dll'
libwdi:debug [AddFileHash] 'libusbk.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\MichaelR1\usb_driver\amd64\libusbk.dll'
libwdi:debug [AddFileHash] 'libusbk_x86.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\MichaelR1\usb_driver\amd64\libusbk_x86.dll'
libwdi:debug [AddFileHash] 'libusb-win32_generic_device.inf': INF type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\MichaelR1\usb_driver\libusb-win32_generic_device.inf'
libwdi:debug [AddFileHash] 'libusb0.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\MichaelR1\usb_driver\x86\libusb0.dll'
libwdi:debug [AddFileHash] 'libusb0.sys': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\MichaelR1\usb_driver\x86\libusb0.sys'
libwdi:debug [AddFileHash] 'libusb0_x86.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\MichaelR1\usb_driver\x86\libusb0_x86.dll'
libwdi:debug [AddFileHash] 'libusbk.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\MichaelR1\usb_driver\x86\libusbk.dll'
libwdi:info [CreateCat] Successfully created file 'C:\Users\MichaelR1\usb_driver\libusb-win32_Generic_Device.cat'
libwdi:debug [CreateSelfSignedCert] Set Enhanced Key Usage, URL and CPS
libwdi:debug [CreateSelfSignedCert] Created new key container
libwdi:debug [CreateSelfSignedCert] Generated new keypair...
libwdi:info [CreateSelfSignedCert] Created new self-signed certificate 'CN=USB\MS_COMP_LIBUSB0 (libwdi autogenerated)'
libwdi:debug [SelfSignFile] Successfully created certificate 'CN=USB\MS_COMP_LIBUSB0 (libwdi autogenerated)'
libwdi:info [SelfSignFile] Added certificate 'CN=USB\MS_COMP_LIBUSB0 (libwdi autogenerated)' to 'Root' and 'TrustedPublisher' stores
libwdi:info [SelfSignFile] Successfully signed file 'C:\Users\MichaelR1\usb_driver\libusb-win32_Generic_Device.cat'
libwdi:info [SelfSignFile] Successfully deleted private key
libwdi:debug [wdi_install_driver] using progress bar mode
libwdi:debug [installer process] got parameter libusb-win32_Generic_Device.inf
libwdi:debug [process_message] got request for device_id
libwdi:debug [process_message] no device_id - sending empty string
libwdi:debug [installer process] got device_id: ''
libwdi:debug [process_message] got request for hardware_id
libwdi:debug [process_message] no hardware_id - sending empty string
libwdi:debug [installer process] got hardware_id: ''
libwdi:debug [installer process] got user_sid: 'S-1-5-21-1324130775-3425519498-3713388002-1001'
libwdi:debug [installer process] using syslog 'C:\Windows\inf\setupapi.dev.log'
libwdi:debug [installer process] syslog reader thread started
libwdi:debug [installer process] successfully disabled the system restore point creation setting
libwdi:debug [process_message] switching timeout to infinite
libwdi:debug [installer process] Copying inf file (for the next time device is plugged) - please wait...
libwdi:debug [process_message] switching timeout to infinite
libwdi:debug [syslog] chaelR1\usb_driver\libusb-win32_Generic_Device.inf]
libwdi:debug [syslog] >>>  Section start 2020/07/05 23:30:28.118
libwdi:debug [syslog]       cmd: "C:\Users\MichaelR1\usb_driver\installer_x64.exe" "libusb-win32_Generic_Device.inf"
libwdi:debug [syslog]      inf: Copy style: 0x00000000
libwdi:debug [syslog]      sto: {Setup Import Driver Package: C:\Users\MichaelR1\usb_driver\libusb-win32_Generic_Device.inf} 23:30:28.120
libwdi:debug [syslog]      inf:      Provider: libusb-win32
libwdi:debug [syslog]      inf:      Class GUID: {EB781AAF-9C70-4523-A5DF-642A87ECA567}
libwdi:debug [syslog]      inf:      Driver Version: 08/02/2012,1.2.6.0
libwdi:debug [syslog]      inf:      Catalog File: libusb-win32_Generic_Device.cat
libwdi:debug [syslog]      sto:      {Copy Driver Package: C:\Users\MichaelR1\usb_driver\libusb-win32_Generic_Device.inf} 23:30:28.126
libwdi:debug [syslog]      sto:           Driver Package = C:\Users\MichaelR1\usb_driver\libusb-win32_Generic_Device.inf
libwdi:debug [syslog]      sto:           Flags          = 0x00000007
libwdi:debug [syslog]      sto:           Destination    = C:\Users\MICHAE~1\AppData\Local\Temp\{f48d08ac-fb75-9f4e-afd7-8552c5d08745}
[... 540 lines cut ...]
[... cut ...]
libwdi:debug [wdi_create_list] Device description: 'USB Receiver (Interface 0)'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_045E&PID_075D&REV_0105&MI_02
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_01&SubClass_02&Prot_00
libwdi:debug [wdi_create_list] Driver version: 10.0.18362.1
libwdi:debug [wdi_create_list] usbaudio USB device (17): USB\VID_045E&PID_075D&MI_02\7&2A616582&0&0002
libwdi:debug [wdi_create_list] Device description: 'Microsoft® LifeCam Cinema(TM) (Interface 2)'

Ho-Ro, 2020-07-06:
Michael,

I see two times an entry for VID_04B5, PID_6022, REV_0000 - this could indicate that the (original supplied by) Hantek driver is still active and blocks OpenHantek. But I do not use Windows at all and cannot check it here. 

    libwdi:debug [wdi_create_list] Hardware ID: USB\VID_04B5&PID_6022&REV_0000
    libwdi:debug [wdi_create_list] Compatible ID: USB\Class_FF&SubClass_00&Prot_00
    libwdi:debug [wdi_create_list] Driver version: 0.0.0.0
    libwdi:debug [wdi_create_list] Hantek6022BE2 USB device (15): USB\VID_04B5&PID_6022\5&2AD60A6B&0&3
    libwdi:debug [wdi_create_list] Device description: 'HantekDSO6022BE'

The hardware of Hantek DSO-6022 mainly consists of a Cypress EzUSB chip, which - when switched on - reports VID/PID 04B4/6022 (these values are stored in the EEPROM). In normal operation OpenHantek claims this USB device and uploads my own firmware into RAM, that finally switches the scope device from VID 04B4 to 04B5, reporting VID_04B5, PID_6022, **REV_0207**. 
I suspect that the original Hantek driver does something similar, but it uploads a different firmware (**REV_0000**) that is missing a lot of features and cannot be used by OpenHantek. They communicate via a closed source dll that is only available for Windows while OpenHantek started 10 years ago to make the scope usable under Linux (the correct Linux/Unix way to communicate with USB is to use libusb). Since OH only used system-independent libraries, it could later be ported to Windows, where a libusb.dll is also available, which must be activated by Zadig.

Please apply Zadig twice:
1. with the freshly plugged in device (VID_04B4, PID_6022)
2. when OpenHantek has loaded its firmware (VID_04B5, PID_6022).

If you do not see the 1st VID/PID combination after plugging the original Hantek has probably grabbed it as soon you plug the device in - in this case please remove the Hantek driver!

Martin

Ho-Ro, 2020-07-13:
No reaction for a week - either the problem is solved or Michael has lost interest. I'm closing for now, please reopen if you're still interested and new information is available.

closed 2020-07-13 by Ho-Ro