OpenHantek/OpenHantek6022 #104: Offset on channel 1 with DSO-6022BL

UweGo, 2020-07-05:
I have an offset error with DSO-6022BL (FW0207) and OpenHantek6022(3.1.1) on Linux. There is no offset error with Hantek software with Windows. 
I short-circuited the probes to ground and readout the DC offset with different settings.
Channel1:
200mV x 10 = 600mV=
500mV x 10 = 600mV=
2V x 10 = 960mV=
5V x 10 = 1,92V=
10V x 10 = 2,83V=
20V x 10 = 2,83V=
50V x 10 = 2,83V=
Channel 2 
200mV x 10 = 278mV=
500mV x 10 = 278mV=
1V x 10 = 278mV=
2V x 10 = 388mV=
5V x 10 = 576mV=
10V x 10 = 297mV=
20V x 10 = 307mV=
50V x 10 = 299mV=
There is no difference in this behavior if I plug the oscilloscope in from time to time.
Compiled on Linux Manjaro 
Linux version 5.4.44-1-MANJARO (builduser@development)  
AMD64
gcc-version 10.1.0 
QT 5.12.0
MESA  20.0.7
Caicos [Radeon HD 6450/7450/8450 / R5 230 OEM]
Start message:
qt5ct: using qt5ct plugin
qt5ct: palette support is disabled
qt5ct: D-Bus global menu: no

Channel 1 and channel 2 connected to GND.
[image]

Channel 1 connected to cal. output.
Channel 2 connected to GND.
[image]

Same settings as before with Hantek software on VM-Virtual-Box.

[image]

Ho-Ro, 2020-07-05:
The Hantek DSO-6022Bx hardware has a quite big offset error due to the [brain-dead HW design](https://github.com/OpenHantek/OpenHantek6022/issues/45#issuecomment-579257234). Therefore it is necessary to calibrate the zero offset, the device provides a factory offset calibration in the [EEPROM](https://github.com/Ho-Ro/Hantek6022API/tree/master/docs#eeprom).

1. Did you [calibrate](https://github.com/Ho-Ro/Hantek6022API/blob/master/README.md#create-calibration-values-for-openhantek) the scope before using it? 
2. Do you have a [calibration file](https://github.com/OpenHantek/OpenHantek6022/blob/8772d08b6556131476bc574c71146913a9af2772/openhantek/src/hantekdso/models/modelDSO6022.cpp#L63)?
3. What is the [content](https://github.com/Ho-Ro/Hantek6022API/blob/master/examples/get_calibration.py) of the scope's calibration EEPROM?
4. Did you calibrate the *Hantek software on VM-Virtual-Box*?

UweGo, 2020-07-06:
(4.) I didn't calibrate the Hantek software on VM-Virtual-Box before. It confused me. The original software didn't show an offset.
(1.) I run this in a terminal:
calibrate_6022.py -e
Setting up scope...

Calculate zero adjustment
Apply 0 V to both channels and press <ENTER> 
Measure offset at low speed for gain  10
Measure offset at high speed for gain  10
Measure offset at low speed for gain  5
Measure offset at high speed for gain  5
Measure offset at low speed for gain  2
Measure offset at high speed for gain  2
Measure offset at low speed for gain  1
Measure offset at high speed for gain  1
eeprom content [  8 .. 39 ]:  b'8f878f878f878c858a838781878187818f878f878f878c858982878087808780'
eeprom content [ 40 .. 55 ]:  b'ffffffffffffffffffffffffffffffff'
eeprom content [ 56 .. 87 ]:  b'837383738373cf613b618845884588456aff6aff6aff91ffb8ff80ff80ff80ff'

After that there is no zero offset with OpenHantek.

closed 2020-07-06 by UweGo

sheerun, 2022-01-26:
could you do calibration in client software instead of eeprom? it's cumbersome to update it

Ho-Ro, 2022-01-26:
@sheerun: please check #189

sheerun, 2022-01-26:
Thank you! Could you please put example .ini file in the readme as I can't easily run get_calibration.py on macos? I think I could write it manually

Ho-Ro, 2022-01-26:
Please have a look at the issue #189, [this comment]( https://github.com/OpenHantek/OpenHantek6022/issues/189#issuecomment-863841296) provides some links to ini files.