OpenHantek/OpenHantek6022 #105: Calibration file problem

spam-receiver, 2020-07-09:
Calibration file "~/.config/OpenHantek/modelDSO6022.ini" alone do not change anything.
Calibration file "~/.config/OpenHantek/modelDSO6022.conf" alone do not change anything too.
But if both these files exist in directiory, then calibration data applies from "modelDSO6022.conf".

Found out under Fedora 32 Linux. Under Windows has not yet tested.

spam-receiver, 2020-07-09:
I could not figure out where to place the calibration file under Windows. Apparently, even all the settings are written to the registry.

closed 2020-07-09 by Ho-Ro

Ho-Ro, 2020-07-09:
Now it should work with an ini file alone, an existing `model*.conf` will be renamed automatically to `model*.ini`.
Regarding the ini file location please check this [Qt doc](https://doc.qt.io/qt-5/qsettings.html#locations-where-application-settings-are-stored):

    Linux/FreeBSD/macOS: $HOME/.config/OpenHantek/modelDSO6022.ini
    Windows: %APPDATA%\OpenHantek\modelDSO6022.ini

commit a71bd67d (2020-07-09): fix #105 - correct ini file handling

https://doc.qt.io/qt-5/qsettings.html#locations-where-application-settings-are-stored
Linux/FreeBSD/macOS: $HOME/.config/OpenHantek/modelDSO6022.ini
Windows: %APPDATA%\OpenHantek\modelDSO6022.ini