OpenHantek/OpenHantek6022 #110: 6022BE Device not found under Windows 10 x64

TaraHoleInIt, 2020-07-31:
**Describe the bug**
Scope does not appear in the drop down list upon launching of the app.
Light next to CH1 remains unlit.

**To Reproduce**
I Installed WinUSB driver using Zadig for VID 04B4 PID 6022.
Scope appears in Device Manager with VID 04B4 PID 6022.

**Expected behavior**
The application to detect the scope.

**Computer environment (please complete the following information):**
 - OpenHantek version: Release 3.1.2 (Tried x32 and x64)
 - OS: Windows 10 x64

**Scope device (please complete the following information):**
 - Device: Hantek 6022BE
 - Program top line: Select a device
 - Input signal: None
 - Probe setting: 1X

**Additional context**
The Hantek drivers and software work, and I completely removed their drivers before attempting to install the new drivers using Zadig.
OpenHantek6022 Just does not see the scope at all.

[image]

Ho-Ro, 2020-08-10:
Hi, 
I see more than 120 downloads of the x64 version and almost 40 of the x86, a positive feedback about version 3.1.2 would be highly appreciated.
From former issues I know @AndreasDerFuchs, @Maddo534, @justlep, @va7ta, @vizkoze and @seraphie21 as active Windows user, maybe one of you can support @TaraHoleInIt.

Thx, Martin

DaPa, 2020-08-16:
@TaraHoleInIt: do you connect the oscilloscope directly to the laptop or via any USB hub(s)?
Sometimes I did have issues connnecting through a chain of hubs (DELL docking station + a USB3 hub), however now I have it connected through the hub and it does work fine

[image]

[image]

AndreasDerFuchs, 2020-08-16:
Maybe you need to install the driver, e.g. with Zadig (from: https://zadig.akeo.ie)
	I used: Zadig - The Automated Driver Installer - Version 2.4 (Build 721)
        and installed (for device DSO-6022BL): WinUSB (v6.1.7600.16385)	

[image]
[image]

DaPa, 2020-08-17:
Indeed, the trick is to run Zadig twice: for both 04B4/6022 and 04B5/6022 devices. Here are my steps:
[OpenHantek6022_zadig_Win10.pdf](https://github.com/OpenHantek/OpenHantek6022/files/5082593/OpenHantek6022_zadig_Win10.pdf)

Ho-Ro, 2020-08-17:
@DaPa: Thank you for this very good tutorial, I would like to include this into the documentation if you don't mind.

The _double-zadig-trick_ is already mentioned (shortly) in [Microsoft Windows USB driver install (with Zadig)](https://github.com/OpenHantek/OpenHantek6022/blob/master/docs/build.md#microsoft-windows-usb-driver-install-with-zadig) of the documentation that is linked on the starting page in section [Run OpenHantek6022](https://github.com/OpenHantek/OpenHantek6022#run-openhantek6022), but your document guides the not so experienced user step-by-step - well done.

DaPa, 2020-08-17:
Please feel free to include the info you need. And I use this opportunity to thank you very much for the great job you are doing with this OpenHantek6022 software!

Ho-Ro, 2020-08-23:
Now the info about the _double-zadig-trick_  and - more important - the link to your tutorial is [available](https://github.com/OpenHantek/OpenHantek6022#windows-usb-access) on the starting page. Thx @DaPa

closed 2020-08-23 by Ho-Ro