OpenHantek/OpenHantek6022 #111: HorizontalDock::setSamplerate startup crash when timebase is 50us or lower (Win10.1909x64)

DaPa, 2020-08-16:
Using latest 3.1.2 app on Windows 10.1909 64 bit with zadig-2.5 drivers.
Have default settings (if not sure, delete HKCU\Software\OpenHantek\OpenHantek6022).
1. Set Timebase to 100us (observe that Samplerate increases to 10MS/s).
2. Restart OpenHantek.exe.
3. Check that app start fine with the previous set settings.
    Double check that in registry [HKEY_CURRENT_USER\Software\OpenHantek\OpenHantek6022\scope\horizontal] there is:
    "timebase"="0.0001"
    "samplerate"="1e+07"
4. Change Timebase to 50us (observe that Samplerate increases to 12MS/s).
5. Restart OpenHantek.exe.
6. Observe that the app starts and crashes immediately.
    Double check that in registry [HKEY_CURRENT_USER\Software\OpenHantek\OpenHantek6022\scope\horizontal] now there is:
    "timebase"="5e-05"
    "samplerate"="1.2e+07"
7. To work around, it's enough to reset registry value "timebase" to "1e-04" and the app will start fine.
8. With VS2019 tracked that the crash happens into HorizontalDock::setSamplerate for a 3rd call when the passed sample rate is 24MS/s (couldn't do it with QT creator 4.12.2 based on Qt 5.14.2 as that was crashing too when the app was crashing...)

[image]
[image]

PS: here are the changes I had to make to get it compilable in VS2019:
https://github.com/DaPa/OpenHantek6022/commit/3b120a3842de63d5c67c6b02018720584f15e463

DaPa, 2020-08-17:
Yes.
[image]
100us works fine with Demo, 50us crashes it.

This reg file revives it:

`
Windows Registry Editor Version 5.00

[HKEY_CURRENT_USER\Software\OpenHantek\OpenHantek6022\scope\horizontal]

"timebase"="0.0001"
`

DaPa, 2020-08-17:
Opening the dump in VS shows this stack:
[image]

`
Qt5Core.dll!00007ffea369310b()
qwindows.dll!00007ffecbd1071f()
qwindows.dll!00007ffecbd0fb57()
UIAutomationCore.dll!00007ffebca2e759()
UIAutomationCore.dll!00007ffebca2f7fc()
UIAutomationCore.dll!00007ffebca2ec9c()
UIAutomationCore.dll!00007ffebca2d744()
UIAutomationCore.dll!00007ffebca2d3fc()
qwindows.dll!00007ffecbd105ba()
Qt5Gui.dll!00007ffea3aab3c2()
Qt5Widgets.dll!00007ffea41d5c47()
Qt5Widgets.dll!00007ffea4226dfa()
Qt5Widgets.dll!00007ffea42261bd()
Qt5Widgets.dll!00007ffea4228f9e()
OpenHantek.exe!00007ff7e4dcba75()
OpenHantek.exe!00007ff7e4dfe84b()
Qt5Core.dll!00007ffea36ae3d1()
OpenHantek.exe!00007ff7e4dc4c05()
OpenHantek.exe!00007ff7e4dec9c6()
OpenHantek.exe!00007ff7e4dec575()
OpenHantek.exe!00007ff7e4deb217()
Qt5Core.dll!00007ffea36b4da4()
Qt5Core.dll!00007ffea36b3b38()
Qt5Widgets.dll!00007ffea40b8810()
Qt5Widgets.dll!00007ffea40b7697()
Qt5Core.dll!00007ffea368ed69()
Qt5Core.dll!00007ffea3690e4d()
Qt5Core.dll!00007ffea36d8935()
user32.dll!00007ffef0765c7d()
user32.dll!00007ffef0765672()
Qt5Core.dll!00007ffea36d8106()
Qt5Core.dll!00007ffea368ac13()
Qt5Core.dll!00007ffea34fede2()
Qt5Core.dll!00007ffea350955d()
kernel32.dll!00007ffeeebe7bd4()
ntdll.dll!00007ffef0aace51()
`

Ho-Ro, 2020-08-17:
Very strange, a quick user-mode installation on a Win10 (unzip into a new folder without installing USB drivers) run well in demo mode with all timings for both versions, x64 and x86. 

    Microsoft Windows [Version 10.0.17763.1339]
    (c) 2018 Microsoft Corporation. Alle Rechte vorbehalten.

DaPa, 2020-08-17:
I confirm your finding: in a VirtualBox with Win10ProN.2004.x64, unzipped OpenHantek-Win-x64-Release-b364.zip and only had to add 3 dlls (msvcp140.dll, vcruntime140.dll and vcruntime140_1.dll: all are version 14.26.28720.3 - copyied from host's C:\Windows\System32) then tested and indeed the demo does work fine with 50us!
[image]

Unfortunatelly as before that same exe is still crashing on my host system... will try to dig more and come up with any extra info.

Ho-Ro, 2020-08-17:
Please check these quick hacked versions https://ci.appveyor.com/api/buildjobs/of3d7bd4pn72siy3/artifacts/build_x64%2Fopenhantek%2FOpenHantek-Win-x64-Release-b365.zip and https://ci.appveyor.com/api/buildjobs/of3d7bd4pn72siy3/artifacts/build_x86%2Fopenhantek%2FOpenHantek-Win-x86-Release-b365.zip that skip the strange _3rd call_ with an unrelated sample rate.

DaPa, 2020-08-17:
Unfortunatelly same result with OpenHantek-Win-x64-Release-b365.zip.
[image]
`>	Qt5Core.dll!00007ffb00c4310b()	Unknown
 	qwindows.dll!00007ffb3c12071f()	Unknown
 	qwindows.dll!00007ffb3c11fb57()	Unknown
 	UIAutomationCore.dll!00007ffb39a3e759()	Unknown
 	UIAutomationCore.dll!00007ffb39a3f7fc()	Unknown
 	UIAutomationCore.dll!00007ffb39a3ec9c()	Unknown
 	UIAutomationCore.dll!00007ffb39a3d744()	Unknown
 	UIAutomationCore.dll!00007ffb39a3d3fc()	Unknown
 	qwindows.dll!00007ffb3c1205ba()	Unknown
 	Qt5Gui.dll!00007ffb0105b3c2()	Unknown
 	Qt5Widgets.dll!00007ffb01785c47()	Unknown
 	Qt5Widgets.dll!00007ffb017d6dfa()	Unknown
 	Qt5Widgets.dll!00007ffb017d61bd()	Unknown
 	Qt5Widgets.dll!00007ffb017d8f9e()	Unknown
 	OpenHantek.exe!00007ff6c325ba8a()	Unknown
 	OpenHantek.exe!00007ff6c328df73()	Unknown
 	Qt5Core.dll!00007ffb00c5e3d1()	Unknown
 	OpenHantek.exe!00007ff6c3254c15()	Unknown
 	OpenHantek.exe!00007ff6c327caac()	Unknown
 	OpenHantek.exe!00007ff6c327d2fc()	Unknown
 	OpenHantek.exe!00007ff6c327c4ca()	Unknown
 	OpenHantek.exe!00007ff6c327b117()	Unknown
 	Qt5Core.dll!00007ffb00c64da4()	Unknown
 	Qt5Core.dll!00007ffb00c63b38()	Unknown
 	Qt5Widgets.dll!00007ffb01668810()	Unknown
 	Qt5Widgets.dll!00007ffb01667697()	Unknown
 	Qt5Core.dll!00007ffb00c3ed69()	Unknown
 	Qt5Core.dll!00007ffb00c40e4d()	Unknown
 	Qt5Core.dll!00007ffb00c88935()	Unknown
 	user32.dll!00007ffb61f15c7d()	Unknown
 	user32.dll!00007ffb61f15672()	Unknown
 	Qt5Core.dll!00007ffb00c88106()	Unknown
 	Qt5Core.dll!00007ffb00c3ac13()	Unknown
 	Qt5Core.dll!00007ffb00aaede2()	Unknown
 	Qt5Core.dll!00007ffb00ab955d()	Unknown
 	kernel32.dll!00007ffb61b47bd4()	Unknown
 	ntdll.dll!00007ffb6398ce51()	Unknown
`
And same behavior ir recovers if I reset "timebase"="0.0001".

OpenHantek-Win-x86-Release-b365.zip is always crashing after pressing Demo Mode and its dump doesn't show valid stack:
[image]
I said that it's always because it doesn't matter if I wipe completely the HKCU\Software\OpenHantek\OpenHantek6022 key. I doublechecked and there is no such key under HKCU\Software\Wow6432Node eighter so the x86 exe might have other troubles on my machine...

DaPa, 2020-08-17:
An extra info. Tested latest 3.1.2 and an older OpenHantek-Win-x64-Release-b356 build on my son's Windows 7 laptop and there they both work fine with 50us!
Then tested again both exes on my daughter's laptop (same Win10.1909.x64.Pro) and there they crashed too: the machines are quite similar (mine: Dell Precision 5530, company's corporate Windows. Her: Dell Precision 5520, own Windows install). So maybe the issue is on 1909? Not sure yet, will come with more updates maybe tommorow.

spam-receiver, 2020-08-17:
> An extra info. Tested latest 3.1.2 and an older OpenHantek-Win-x64-Release-b356 build on my son's Windows 7 laptop and there they both work fine with 50us!
> Then tested again both exes on my [...]

I have Windows 10 Pro x64 1909 18363.1016 and there is no such strange issues. It may be related to some Dell's software or something else.

DaPa, 2020-08-17:
Thank you @Ho-Ro, 94c1f15 did it, now every timebase below 100us works! I just built it with VS2017 and works fine every restart.
[image]

DaPa, 2020-08-17:
Closing the issue! Here's few tests with real device and a real SPI signal. After each restart the app remebers the low timebase and works fine:
[image]
[image]

DaPa, 2020-08-18:
Both OpenHantek-Win-x86-Release-b368 and OpenHantek-Win-x64-Release-b368 work fine too!

[image]

closed 2020-08-18 by DaPa