OpenHantek/OpenHantek6022 #113: Ability to set font size via .conf

technosf, 2020-08-27:
OpenHantek UI starts up with an **_unreadably_** small font on my hw/s/w setup.

Unable to see any menu options to set font size, I would like to have font config options in the OpenHantek6022.conf so I can up date the settings. 

Settings would be font and font size

TY!

Ho-Ro, 2020-08-27:
> on my hw/s/w setup

Please provide an useful amount of information about your setup - e.g. hardware, OS, distribution, screen resolution, maybe a screen shot, ...

technosf, 2020-08-27:
Linux Mint 
[image]

5.4.0-42-generic #46-Ubuntu SMP  x86_64 x86_64 x86_64 GNU/Linux
2736x1824 (3:2), 60hz, 2x scale -> 1368x912 scaled resolution

Ho-Ro, 2020-08-27:
This is not a recent OpenHantek6022 version - isn't it? Where did you get it?

The 1st screen looks different since about one year.

[image]

technosf, 2020-08-27:
Installed from the 3.1.2 release file: openhantek_20200729-95ff65f_amd64.deb

Ho-Ro, 2020-08-27:
> openhantek_20200729-95ff65f_amd64.deb

Nope!

Please provide the output of 
`OpenHantek -v`
and  
`which OpenHantek`

I guess you have an old install elsewhere, e.g. in `~/bin` or `/usr/local/bin`

technosf, 2020-08-27:
Install is good:
```
/usr/bin/OpenHantek
OpenHantek6022 3.1.2
```
I think I've found the issue - its Qt5 interacting with the display scaling - Using the display magnifier and the Qt5 settings app I knocked the Qt5 fonts up into the 30point range and it is now close to your image above.

Thank you for your quick attention to this ticket and for you work on OH6022!

closed 2020-08-27 by technosf

Ho-Ro, 2020-08-27:
One question, where does this tabbed window in your screen shot come? It is no more in OH6022, I removed the tabs about one year ago. OH6022 also does not support only DSO-6022BE and DSO-6022BL, but also some other compatible devices, as well as a DEMO mode (as seen in my screen shot above).

[image]

technosf, 2020-08-28:
The font setting does not affect the font size. I tried with "-f" and "--fontsize" with various values. 

This was in the terminal on running:
_qt5ct: using qt5ct plugin_

Not sure about the tabs - they are not there now. I keep an eye out to see if I recreate that.

Ho-Ro, 2020-08-29:
You can try these options: https://doc.qt.io/qt-5/highdpi.html#high-dpi-support-in-qt

technosf, 2020-08-30:
My QT:
_QMake version 3.1
Using Qt version 5.12.8_

QT_SCALE_FACTOR: scales the font and the layout, so OpenHantek is too big to be usable, and the font still hard to read without the chang of fonts size in qt5ct.conf. 

The other env. variables had no effect on on the layout scaling nor the font rendering.