OpenHantek/OpenHantek6022 #117: OpenHanotek hangs in macOS 11.0

sergiocampama, 2020-09-05:
**Describe the bug**
After opening the .app, it hangs forever. Attached is a spindump of the process

**To Reproduce**
**IMPORTANT:** Always unplug and replug the scope if used before with a different application, e.g. sigrok or the original Hantek software.

Steps to reproduce the behavior:
1. Download version 3.1.2 for macOS on a macOS 11.0 installation.
2. Open it

**Expected behavior**
Expected app to open

**Screenshots**
App doesn't open

**Computer environment (please complete the following information):**
 - OpenHantek version: 3.1.2 release from releases
 - OS: macOS 11.0 beta (20A5364e)
 - Distribution, version: MacBook Pro 16"

**Scope device (please complete the following information):**
 - Device DDS120 (but app hangs with the device connected or not)

**Additional context**
Add any other context about the problem here, e.g. the program output if started in a console like xterm.

[sample.txt](https://github.com/OpenHantek/OpenHantek6022/files/5177718/sample.txt)

Ho-Ro, 2020-09-10:
@sergiocampama  Did you try to compile it on your system?

Ho-Ro, 2020-10-04:
I have the feeling that you are no longer interested in this subject - will close now.

closed 2020-10-04 by Ho-Ro