OpenHantek/OpenHantek6022 #119: Allow high samplerate also with longer timebase

TPReal, 2020-09-17:
**Is your feature request related to a problem? Please describe.**
I have some signal I want to observe. The signal, from start to end, takes about 2ms. I'm interested in something at the very end of the signal. So I set the trigger so that the start of the signal triggers, and I need to set the timebase to 200 microseconds to fit the whole signal on screen. Then I zoom in at the end of the signal - and the time resolution that I see is too low. This happens because with timebase of 200 micros, sample rate is 5MS/s, and cannot be made higher in the UI. What's even worse in my case, downsampling is done, so if my signal's real frequency is high enough, I see various artifacts in the zoomed in view.

**Describe the solution you'd like**
I'd like to be able to always select the effective samplerate as high as I want, even if it disables the downsampling, so that I can get high time resolution on zoomed in view.

**Describe alternatives you've considered**
What I'm currently doing is I set lower timebase and then "hunt" for the end of the signal trying to pause at the right time. Or put an auxiliary signal on the second channel that I use as a trigger.
An alternative feature that would help me is setting a time offset for the trigger, equivalent to moving the trigger point (on the time axis) left, far beyond what's on screen. For example I'd then set timebase to 5 microseconds, maximum samplerate, and say that I want the graph to start 1.95ms after the trigger occurs. This way I'd have full time resolution of the last 50 micros of my signal.

**Additional context**
I'm using OpenHantek on Windows.
Screenshot of the situation, see how the zoomed in part has very low time resolution. I can get a much nicer view of the "packet" if I set the timebase lower, but then I cannot easily see the LAST packet.
[image]

Ho-Ro, 2020-09-22:
> I'd like to be able to always select the effective samplerate as high as I want, even if it disables the downsampling, so that I can get high time resolution on zoomed in view.

TL;DR: This will not happen.

If you look at the [data flow](https://github.com/OpenHantek/OpenHantek6022/blob/master/docs/readme.md#data-flow) you will see that the software takes raw values (8bit) with various sample sizes. These raw byte values are converted to physically meaningful double values (measured in Volt) with constant number of samples (20K samples per channel). All further processing is done on these blocks of constant size. This size is a trade-off between enough details and calculation time. The software trigger uses half of the buffer size to show a stable trace on the screen (it finds a 10K Sample block out of 20K samples, so that the trigger position is always aligned on the same screen position). The spectrum postprocessing (FFT) uses the complete 20K block to get the best accuracy. The FFT ist always active - even when the spectrum is not displayed - because is delivers the signal frequency displayed on the bottom.
We have two timing/processor load bottlenecks, 1st the FFT, 2nd the OpenGL trace update. Especially the display of 100K samples on screen doesn't make sense on normal screens that are about 1000..2000 dots wide.

> An alternative feature that would help me is setting a time offset for the trigger...

Your suggestion to introduce a trigger delay does not work because the device has no HW trigger, but blindly reads a data block asynchronously and then searches for a trigger condition in this block.
What I did in former single-channel-analog-scope times: get an auxiliary signal , e.g. start of frame and delay it with an external time-variable monoflop (we had some battery powered boxes, 74LS123 or 74LS221, TTL in, TTL out in the lab) and feed it into the external trigger input of the scope. In your case you can use the 2nd channel.

Sorry for the rejection, but you hit the limitations of this cheap HW.

Another possibility (for lower sample rates up to 100 kS/s) would be to use the Hantek as a
[data logger](https://github.com/Ho-Ro/Hantek6022API#use-the-device-as-a-data-logger) and post-process the data offline, e.g with a spreadsheet program or your own python script.

closed 2020-09-22 by Ho-Ro

TPReal, 2020-09-25:
Thank you for the detailed explanation. So if I understand correctly, it's
not impossible because of the hardware limitation, but rather it's
impossible with the current architecture of the software, and changing this
architecture (like, adding an option to increase the block size) is just
not viable.

Is there any reason why the data logger cannot use the higher sample rates?
It looks like neither of the bottlenecks you mentioned applies to a data
logger.

As for setting the time offset for the trigger - this still sounds doable
somehow. Currently in normal trigger mode I guess you scan each block
looking for the trigger, if it occurs then you display this block (aligned
to the trigger), and if not, you ignore that block. My suggestion would
mean that you make the decision whether to display or ignore the block
based on the appearance of the trigger not in this block, but in the block
n blocks ago. This requires storing a rolling window of n booleans (whether
i'th block contained a trigger) in the minimal implementation. For 20MS/s
sampling, one second is just 1000 blocks, if I count correctly, and storing
1000 of booleans (that you computed anyway) definitely sounds doable. Or
even storing the exact time of the trigger for each of the past blocks, to
allow aligning of the display to the exact delay after trigger.

I solved my problem in some other way (similar to your suggestion about the
auxiliary signal), but I still believe my suggestion is valid and could be
a nice additional feature of the software. Would you mind considering it
again?

Ho-Ro, 2020-10-02:
> Is there any reason why the data logger cannot use the higher sample rates? It looks like neither of the bottlenecks you mentioned applies to a data logger.

If you look at [this part](https://github.com/Ho-Ro/Hantek6022API/blob/af7aa26526cb2d8801862a8ade3b56666b6dadaf/examples/capture_6022.py#L107) the data is saved in a callback between blocks, if the blocks come in to fast an overrun happens with data loss. 

> Would you mind considering it again?

This would require that the n blocks are aligned back to back, but unfortunately there are undetermined gaps in between, so you cannot delay the trigger by a defined time.
Sorry, but nevertheless, it would have been a clever solution - thx for the proposal.