OpenHantek/OpenHantek6022 #120: Instrustar ISDS205* support

22224444, 2020-09-25:
Program will work with ISDS205X scope if add model files like modelDSO6022.cpp and modelDSO6022.h with modified vid/pid of no FW state of scope (for example 0x04b4, 0x6022 replace to 0xd4a2, 0x5661) and add include of modified file to hantekdsocontrol.cpp. Program will upload firware from DSO6022 and scope will work with 2 channels. Unfortunatelly I cannot calibrate scope for work with 1x/10x probe divider. Also DDS generator and logic analyzer not works and ac/dc switch not working (maybie something else will not work or work wrong). Can I ask how I can dump firmware from ISDS205X using wireshark (or something else)? Try to do this using wireshark but found not all data (for example from firmware of DSO6022 in second packet I found first string (but not full) 0400 - 36, 37 byte, 0000 - 42, 23 byte, 02005932 - data but I cannot find 6F from end of first string. Also I cannot find :2000 from 15 string of firmware. Where I can find more info? what I should read?

Ho-Ro, 2020-10-02:
To start with the (probably) easy task - the LA should work with sigrok or the salae sw.
The difficult part would be the firmware for scope/DDS. If you look at https://sigrok.org/wiki/Instrustar_ISDS205X and https://sigrok.org/wiki/Instrustar_ISDS205A you could draw rough schematics, compare it with the 6022BE schematics and (if you are patient) roll your own firmware based on e.g. my [6022 work](https://github.com/Ho-Ro/Hantek6022API) - or unite your effort with the sigrok team. 
You could also read the [Big-6022-thread](https://www.eevblog.com/forum/testgear/hantek-6022be-20mhz-usb-dso/) from down-under to find out how to grep the FW or use the cypress tools - upload fw withe win sw and read out the FX ram.
Good luck!

Ho-Ro, 2020-11-09:
No reaction since 1 month - closed!
Feel free to reopen if you want to contribute.

closed 2020-11-09 by Ho-Ro

wakass, 2023-10-14:
I've added initial support for Instrustar here: https://github.com/wakass/OpenHantek6022/commit/59d7aa3c911b218ecfb2a8c1acf00f1f63307dd8

- Adds support for Instrustar 205A/B
- Uses fx2lafw (slightly modified to support calibration) firmware from sigrok project
- No fine calibration done yet

Gain settings are a bit dubious for the lower ranges due to lack of a proper low-voltage signal generator. It at least gave ballpark the same values throughout all ranges for a hacked together low voltage source (~200m Vpp). My theoretical calc of 4096 for the 16x amp range (with /10 attenuator disabled) is quite different from the eventual value. So i'm not sure what i'm doing wrong.

Will make a pull-request once the values are dialed in nicely.

xdevelnet, 2025-07-14:
@wakass will it work with 205C?