OpenHantek/OpenHantek6022 #122: Connection Failed! With proper installed WinUSB driver on both 04B4 and 04B5

Kashouryo, 2020-10-04:
**Describe the bug**
The software can not connect the scope in a normal way. If I start the software first then plug the scope in after 5 sec, I can select the scope in the drop-down and enter the software(Have to be fast before the button grey-out). After that, the software works as expected.

**To Reproduce**
Plug the scope into the USB port, starts the software, error.

**Expected behavior**
Starts up normally

**Screenshots**
[image]

**Computer environment (please complete the following information):**
 - OpenHantek version: 16a52e8
 - OS: Windows 10
 - Distribution, version: 2004
 - Video hardware: Not useful for this issue
 - OpenGL version: Not useful for this issue
 - Qt version: Not useful for this issue
 - C++ compiler version: Not useful for this issue

**Scope device (please complete the following information):**
 - Device Hantek 6022BL
 - Program top line Unable to start
 - Input signal: N/A
 - Probe setting: N/A

Ho-Ro, 2020-10-04:
Please check in the device manager which other SW grabs the device, I suppose you have the original Hantek driver still on your system. OpenHantek cannot coexist.

Kashouryo, 2020-10-04:
This is a fresh Windows installation. OpenHaktek6022 is the only Hantek software that I want to use.

DaPa, 2020-10-04:
@Kashouryo  Do you connect the ossciloscope directly to your laptop/PC or via any USB hub(s)? I had issues with devices discovery when they were connected through my DELL docking station when a newer ASMedia driver was installed instead of older Microsoft one (in my case had to choose rollback driver for the "ASMedia USB 3.0 eXtensible Host Controller").
Also do you have this issue wit holder 1909 version of Windows 10 (I have 1909, 64-bit)?

Kashouryo, 2020-10-04:
I am directly connecting my device to the back of my desktop pc. This is not an issue on my 1903 workstation.

Kashouryo, 2020-10-04:
Interestingly... It works with a USB hub on my 2004 desktop...

bcrowell, 2020-10-15:
[off topic]I'm using OpenHantek with my electricity and magnetism classes this semester at Fullerton College. Thanks so much for making this available as open source, which is really helping us a lot in maintaining a good-quality lab curriculum while our students are quarantined at home. [/off topic]

Most Windows users (i.e., most of my students) are getting this error. It's hard for me to analyze or debug, since I don't have a windows machine and I'm not in the room with them. However, it seems like the workaround for this problem is that they have to install the USB driver using Zadig, start OpenHantek, it fails with this error, and then they go through the cycle one or more additional times (re-running Zadig, restarting OpenHantek). The first time they run Zadig, it says something like "undefined device," but the second time they run it, it knows what the device is.

On a possibly related issue, a mac user reports that she succeeded in running OpenHantek only by just re-running the software repeatedly until it worked.

I don't think this is a problem where they've previously used the manufacturer's Windows software. I never called their attention to that software's existence, and many of them don't even have CD drives.

Ho-Ro, 2020-10-16:
You have to run Zadig twice - 1st time for the device without firmware (USB VID/PID are 0x04b4, 0x6022). OH then detects it via libusb as a Hantek6022 and uploads the firmware (this changes the USB VID/PID to 0x04b5, 0x6022). Then you have to run Zadig  the 2nd time to setup libusb also for this device to make it usable.
There is also a signed WinUSB driver for the Hantek 6022BE/BL available at the [eevblog forum](https://www.eevblog.com/forum/testgear/hantek-6022be-20mhz-usb-dso/msg3218116/#msg3218116), but due to the lack of any windows machine I did not test it.

bcrowell, 2020-10-17:
I see, thanks. The tutorial by DaPa actually seems pretty clear. I think my students just weren't reading it.

closed 2020-11-09 by Ho-Ro