OpenHantek/OpenHantek6022 #127: Noise gate on trigger

alpytron, 2020-11-07:
It would be a nice feature if the trigger could apply some filtering to make it more reliable on noisy signals.

Commercial oscilloscopes have such feature, so I guess it won't be too difficult to implement.

Below is a simple example what happens when there is a little amount of noise on the signal. Note that trigger slope is set to rising edge, still sometimes it triggers on falling edge as well.
[image]

Ho-Ro, 2020-11-07:
Please try the trigger setting "smooth":

[image]

alpytron, 2020-11-08:
Thanks for the advice, I was not aware of that item in the Source list. 
I tested it, and it woks quite, however sometimes I still got unstable triggering:
[image]

Maybe it would be good to expose the smoothing/filtering parameter(s) to the user somewhere. Is that something feasible?

closed 2020-11-09 by Ho-Ro

Ho-Ro, 2020-11-09:
Binaries of this version are available as latest [unstable release](https://github.com/OpenHantek/OpenHantek6022/releases/tag/unstable).

alpytron, 2020-11-12:
Thank you, Ho-Ro!
Selecting LF works perfectly for low frequency noisy signals. Great job! :)

commit a012419a (2020-11-09): HF/Normal/LF trigger, fixes #127