OpenHantek/OpenHantek6022 #128: chrash on upgrade from 3.1.3 to unstable 202001109

vizkoze, 2020-11-12:
**Describe the bug**
Updating on ubuntu budgie 20.04 with dpkg -i over existing install crashes the program.

**To Reproduce**
**IMPORTANT:** Always unplug and replug the scope if used before with a different application, e.g. sigrok or the original Hantek software.
Did it several times

Steps to reproduce the behavior:
1. '.download latest openhantek_20200930-16a52e8_amd64.deb file form github.'
2. sudo dpkg -i openhantek_20200930-16a52e8_amd64.deb
3. open program, crash on program
Downgrade to latest stable version restores program in working state. (I always update over existing install and worked in the past)

**Expected behavior**
No crash on update

**Screenshots**
If applicable, add screenshots or a video to help explain your problem.

**Computer environment (please complete the following information):**
 - OpenHantek version: 20201109-a012419
 - OS: linux 
 - Distribution, version: ubuntu budgie 20.04 amd64bit kernel 5.4.0-52-generic
 - Video hardware:I ntel® HD Graphics 3000 (SNB GT2)
 - OpenGL version: *OpenGL 3, Mesa 20.0.8
 - Qt version: [5.12.8 (x86_64-little_endian-lp64 shared (dynamic) release build
 - C++ compiler version: gcc 9.3.0

**Scope device (please complete the following information):**
 - Device  *Hantek 6022BE, 
 - Program top line [e.g. *OpenHantek6022 (Build ...) - Device ... (FW ...)*]
 - Input signal: open
 - Probe setting: [10x

**Additional context**
When starting from commandline this error appears:
qt5ct: using qt5ct plugin
qt5ct: D-Bus global menu: no
/usr/include/c++/7/bits/stl_vector.h:815: std::vector<_Tp, _Alloc>::const_reference std::vector<_Tp, _Alloc>::operator[](std::vector<_Tp, _Alloc>::size_type) const [with _Tp = double; _Alloc = std::allocator<double>; std::vector<_Tp, _Alloc>::const_reference = const double&; std::vector<_Tp, _Alloc>::size_type = long unsigned int]: Assertion '__builtin_expect(__n < this->size(), true)' failed.
Aborted (core dumped)

Ho-Ro, 2020-11-12:
Thx for reporting! Please delete the config file and try again. Or even better, select trigger source CH1 or CH2 (without smooth) in stable and upgrade again. I will also check accordingly.

vizkoze, 2020-11-12:
I tried closing the stable program with trigger source to CH1 (without smooth) and upgrading but that crashed the program like before.
Renaming the config file to OpenHantek6022.conf.backup and restarting the program fixed the issue. (didn't have to reinstall the latest unstable version. A new configfile was generated with default settings.
Would you like a copy of my old config file to see what could be wrong?
I guess re-applying all my settings will be the next thing to do.

Ho-Ro, 2020-11-13:
Yes, that was an incompatible change - I did implement a [config file version value](https://github.com/OpenHantek/OpenHantek6022/blob/a012419affffb478d408a422621aa70d26195fe6/openhantek/src/dsosettings.h#L3) long time ago but forgot to increment its value :(

    // increment this value after incompatible config changes
    #define CONFIG_VERSION 1

A new unstable will come soon - stay tuned!

commit 163da3f7 (2020-11-13): fix #128 - incompatible config file change due to new smooth mode
closed 2020-11-13 by Ho-Ro