OpenHantek/OpenHantek6022 #129: Dummy load power show

bmsorin, 2020-11-14:
Hello,

Because i mainly work on audio projects, i consider to be a good addition to show the power RMS on the main dock.
I have made this modification it look something like this:
[image]
[image]

[patch-3.1.3.diff.zip](https://github.com/OpenHantek/OpenHantek6022/files/5540423/patch-3.1.3.diff.zip)

Attached the diff file for v3.1.3 if someone is interested.

Regards!

Vascom, 2020-11-14:
How you calculate the power?

bmsorin, 2020-11-14:
(Vrms*Vrms)/dummy load

bmsorin, 2020-11-14:
i use this formula
(analysedData.get()->data( channel )->rms*analysedData.get()->data( channel )->rms)/view->dummyLoad

Vascom, 2020-11-14:
I understand. Thanks.
Seems as good improvment. Can you make Pull Request?

bmsorin, 2020-11-14:
Unfortunately i can't make a pull request.
I have take the master branch, create new branch locally, apply the pacth, check all is ok, but when i want to make a push with new branch, i got permission denied, can you tell me what i make wrong?

$ git branch -a
* dummy_load
  master
  remotes/origin/HEAD -> origin/master
  remotes/origin/cleanup_6022
  remotes/origin/last_legacy_support
  remotes/origin/master

$ git push -u origin dummy_load
ERROR: Permission to OpenHantek/OpenHantek6022.git denied to bmsorin.
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.

Regards!

Vascom, 2020-11-14:
You need clone the repo at github. Make your changes. Press button "Create Pull Request".
It is github feature, not git.

bmsorin, 2020-11-14:
ok, here you go
https://github.com/OpenHantek/OpenHantek6022/pull/130

Ho-Ro, 2020-11-14:
Hi, good idea. As a small change I would move the resistance setting from scope into the next tab (currently named  *spectrum*, renamed to *analysis* with subsections *spectrum* and *power*). I will check ..

Ho-Ro, 2020-11-14:
Thx, I merged your #130 with the changes mentioned above.

closed 2020-11-14 by Ho-Ro