OpenHantek/OpenHantek6022 #131: Upgrading MacOS to Big Sur causes application to stop working

bcrowell, 2020-11-18:
Two of my students upgraded their macs to MacOS Big Sur, and they report that afterward, the OpenHantek6022 application no longer worked. When they double-click on the icon, it doesn't open a GUI window on the screen, and the application doesn't respond. The CPU seems to max out (laptop gets hot, fan comes on), and they have to force-quit the application.

Ho-Ro, 2020-11-19:
I cannot support because I develop and test only for Linux. I do not have any Mac nor Win computers to check.
This cannot be solved unless a volunteer steps in - sorry.

bcrowell, 2020-11-19:
Yeah, I understand, thanks. Hopefully a volunteer will step up.

By the way, my school is near the end of our first semester of using this software for physics labs that students do at home during the covid epidemic. The software has worked pretty much flawlessly for our purposes. Thanks for all your work on it!

wrutter-de, 2020-11-22:
I got the same problem, in between i as able to build a version on my Mac with Big Sur (find the app attached).
I was able to build it the official build instructions found here:
https://github.com/OpenHantek/openhantek/blob/master/docs/build.md#apple

[OpenHantek.zip](https://github.com/OpenHantek/OpenHantek6022/files/5579971/OpenHantek.zip)

Ho-Ro, 2020-11-22:
Ok, looks like a build on *Big Sur* will also run on *Big Sur*. 
It would be interesting to know if your *Big Sur* build also works for older versions - positive/negative feedback from MacOS users would be very welcome.
Unfortunately my CI tool [travis](https://travis-ci.com/github/OpenHantek/OpenHantek6022) doesn't provide a *Big Sur* build environment [at the moment](https://docs.travis-ci.com/user/reference/osx/).

Ho-Ro, 2020-11-22:
ok, *Catalina* [doesn't build the *.dmg](https://travis-ci.com/github/OpenHantek/OpenHantek6022/jobs/445438010) - next try ...

Ho-Ro, 2020-11-22:
[same for Mojave](https://travis-ci.com/github/OpenHantek/OpenHantek6022/jobs/445441441) ...

wrutter-de, 2020-11-23:
Tried tu run the Big Sur build on High Sierra, i got a message that the application needs OSX 11.0.
Would be to easy if one fits all ;-)

Ho-Ro, 2020-11-23:
Honestly speaking I expected this kind of missing backward compatibility.

But nevertheless I struggle also with `*.dmg` build issues for *Catalina* as mentioned above:
`macdeployqt` and `macdeployqtfix.py` run with success, but `create-dmg` returns with error:

    if test -d openhantek/OpenHantek.app; then 
        export PATH=${CMAKE_PREFIX_PATH}/bin:${PATH} ; 
        cd openhantek ; 
        macdeployqt OpenHantek.app -always-overwrite -verbose=1 ; 
        python ../../utils/macdeployqtfix/macdeployqtfix.py 
            OpenHantek.app/Contents/MacOS/OpenHantek ${CMAKE_PREFIX_PATH} ; 
        create-dmg --volname OpenHantek --volicon ../../openhantek/res/images/openhantek.icns 
            --window-pos 200 120 --window-size 800 400 --icon-size 100 --icon "OpenHantek.app" 200 190
            --hide-extension "OpenHantek.app" --app-drop-link 600 185 --eula ../../LICENSE 
            OpenHantek.dmg OpenHantek.app ; 
        cd .. ; 
    else
        ls -l packages ;
        true;
    fi
    INFO | fixing executable 'OpenHantek.app/Contents/MacOS/OpenHantek'
    ...
    INFO | macdeployqtfix terminated with success
    Creating disk image...
    created: /Users/travis/build/OpenHantek/OpenHantek6022/build/openhantek/rw.OpenHantek.dmg
    Mounting disk image...
    Mount directory: /Volumes/OpenHantek
    Device name:     /dev/disk2
    Making link to Applications dir...
    /Volumes/OpenHantek
    Copying volume icon file '../../openhantek/res/images/openhantek.icns'...
    Running AppleScript to make Finder stuff pretty: /usr/bin/osascript             
    "/var/folders/z3/_825pg0s3jvf0hb_q8kzmg5h0000gn/T/createdmg.tmp.XXXXXXXXXX.fmWG13ZA" "OpenHantek"
    /var/folders/z3/_825pg0s3jvf0hb_q8kzmg5h0000gn/T/createdmg.tmp.XXXXXXXXXX.fmWG13ZA:396:408: execution error: 
    Finder got an error: AppleEvent timed out. (-1712)
    Failed running AppleScript
    "disk2" ejected.

It has been a long way to MacOS bundles, see #25. Now I have to double check with the original `xcode9.4` setup and where the  regression starts - but unfortunately my travis builds have been disabled temporarily due to missing build credits, obviously I've triggered too many builds yesterday.

Ho-Ro, 2020-11-25:
Now I've switched completely to [appveyor](https://ci.appveyor.com/project/Ho-Ro/openhantek6022) as CI tool that provides Linux, Win and Mac builds.
A new *Catalina* build is now available in [unstable](https://github.com/OpenHantek/OpenHantek6022/releases/tag/unstable).

wrutter-de, 2020-11-27:
Good Job! 
The new Build works under Big Sur.

closed 2020-11-27 by Ho-Ro