OpenHantek/OpenHantek6022 #132: THD measurement

zen67, 2020-11-23:
Hi!
Great work, big thanks!
Would it be possible to implement a THD measurement (audio amplifiers, etc.)?

closed 2020-11-30 by Ho-Ro

Ho-Ro, 2020-11-30:
Enable the THD measurement in menu Oscilloscope/Settings/Analysis. It calculates
 `THD = sqrt( power_of_harmonics / power_of_fundamental )`
Please keep in mind that due to the low-cost HW and the 8 bit ADC a THD < 0.5 % cannot be measured.

zen67, 2020-12-01:
Thank You very much!
Yes, HW is crap.
I'm gonna try to make some improvements (like I'm usually do with audio/digital equipment), to squeeze everything possible.
Best regards!

Ho-Ro, 2020-12-01:
Did you read the [monster thread](https://www.eevblog.com/forum/testgear/hantek-6022be-20mhz-usb-dso/) from "down under"? It shows some [HW mods](https://www.eevblog.com/forum/testgear/hantek-6022be-20mhz-usb-dso/msg344680/#msg344680).

zen67, 2020-12-01:
Yes, II did.
But problem is more complicated.
RichardK did the shielding, but actually, problem is in the common ground plane, which is shared by AD converter, analog input stage and CY7C68013A. More then that, AD converter and CY7C68013A share one 3.3 PSU, which is unacceptable.
Pulse power supply for analog input stage is unacceptable too.
I'll try to separate ground planes, implement extra power supply 3.3 for AD (I have already split the board for this) & analog VCC of CY7C68013A.
More than this I plane to try a USB isolation by ADuM4100/3100 (OWON VDS1022I uses one for USB isolation), make separate power supply +/-5V to power AD8065 (which is noisy actually).
If I achieve some success, I'll write down all the mods in the tread (monster tread :D).
This unit is made, like a toy... just like another Chinese thing - Topping D10 DAC, which is heavily moded to bring some sound out.

Ho-Ro, 2020-12-02:
> Yes, I did. Thanks.

I see. I didn't visit eevblog for a longer time - you have a nice collection of data!

> This piece of hardware is designed very badly, very badly.
> No one thought when they designed the hardware.

Also [this](https://github.com/OpenHantek/OpenHantek6022/issues/45#issuecomment-579257234) is a no-go. The input voltage of the ADC (Ain=-0.5V..+0.5V, /Ain=0) is far outside the specification (common mode voltage: +0.7V..+1.1V). I saw that you thought about the AD8138 - in the end it would be easier to buy a better scope. 
But nevertheless I had the same intentions when I work at OH6022 - squeeze as much as possible out of the impossible :)

The easiest solution for this redesign would be to use a cheap [EzUSB board](https://sigrok.org/wiki/Lcsoft_Mini_Board) as digital backend and build a separate analog front end. It will run mostly without SW change, only the gain and AC/DC switching must move from port C and E to port A, which would lead to a new firmware.
We need 1 calibration out, 2x gain CH1, 2x gain CH2, 1 AC CH1, 1 AC CH2 and one LED(?) (=8 bit)
This could also give the opportunity to implement better gain steps than the 4 original ones (1,2,5,10), e.g. 0.5, 2, 10, 50 or even 0.1, 1, 10, 100 with better resolution in the low level while also allowing to measure up to 10..20 V directly without clipping.

Just my 2 ¢

Martin

> P.S.: The lower timebase/samplerate is, the smaller (low level) popcorn is.

I implemented massive oversampling (up to 200x) to get down to 100 Samples/s while the min samplerate from firmware is 20 kS/s, please have a look at the [source](https://github.com/OpenHantek/OpenHantek6022/blob/a51706cc7d39b98e55c34c1046df552bda38a407/openhantek/src/hantekdso/models/modelDSO6022.cpp#L101).
You probably know [this AN118](https://www.silabs.com/documents/public/application-notes/an118.pdf). Even if we do not have a real white noise overlay it yields some dB of SNR which is best seen in the spectrum display.

zen67, 2020-12-02:
Hi!
Thanks!
You did an exceptional job! Congratulations!

If we add another AD8065 like schematics below
https://drive.google.com/file/d/1xTVkzNH7xWDjiyhF5YO3EPKAUaViWQuW/view?usp=sharing
and use AD converter properly ie full deferential input, replacing 1k with 2k in the input stage (second AD8065) to "normalize" gain? We need clean analog ground plane anyway if we want to remove "popcorn" (I hope).
I want to try to squeeze maximum, with no much additionlal costs (if possible), so more people can make these mods (if they want of course).

Can You comile a version for Windows without restriction of samplerate (48MHz)?
This is for testing purposes.

Ho-Ro, 2020-12-02:
> USB isolation by ADuM4160/3160

This will not work as the data sheet tells: *Low and full speed data rate: 1.5 Mbps and 12 Mbps* and Hantek6022 uses High speed 480 Mbps.

> Can You comile a version for Windows without restriction of samplerate (48MB)?

32 and 64 bit versions for testing (available for 6 months):
https://ci.appveyor.com/project/Ho-Ro/openhantek6022/build/job/xjmv4nry31a7ad3v/artifacts

zen67, 2020-12-02:
Super!
Thanks, for the build. This was quick!
Schematics:
https://flic.kr/ps/3V1i2X

Yes, ADuM4160 is slow. OWON in other hand use it + additional power isolator.
I have already made an external 5V line power supply and I will try this ADuM4160, whether it will work, despite the requirements declared by Hantek. Hit or miss :)

If You use LTSpice, here (available for 1 month) are sim files for schematics above
https://dox.abv.bg/download?id=d3c91ebb31#

P.S. Schematic of AD8065 is mess, but I was curious to see what behavior such a configuration will have. After all AD8065 is cheap :)

Ho-Ro, 2020-12-02:
> Yes, ADuM4160 is slow. OWON in other hand use it

... because OWON is USB 1.1 while Hantek is USB 2.0 high speed, this is described in [EEVBLOG]():

*"Where the Hantek just uses an EzUSB micro for interfacing the ADC to USB, the Owon includes a Silicon Labs SiM3U156 32 bit ARM Cortex-M3 CPU together with a Xilinx Spartan-3 FPGA. There's clearly a lot more happening on-board than in the Hantek. Presumably the triggering is all handled on board (especially as there is a Trigger-out), together with data compression needed to pipe the data via USB1.1 at decent refresh rate. It also results in very low PC overhead with minimal impact on running other applications even on a low-end PC. The 5k memory depth is clearly a less desirable side effect.
... 
The USB 1.1 interface also allows the USB isolation, which is performed by an ADUM3160. USB power is isolated by a TI DCP020505 2W isolated DC-DC converter. This has the lower isolation voltage of the two devices at 1kV."*

zen67, 2020-12-02:
Seems I have missed this post...
In that case, my DAC will be happy to know what awaits it :)
Thanks for the info!
However, nothing prevents us from trying, the widget has already been ordered.
Is there a way to measure the transfer between Hantek and PC?

Ho-Ro, 2020-12-03:
> Is there a way to measure transfer speed between Hantek and PC?

The scope uses high speed 480 Mbps bulk transfer, defined by the [firmware](https://github.com/Ho-Ro/Hantek6022API/tree/master/PyHT6022/Firmware). 
This results in a theoretical (gross) data rate of 60 MS/s, the maximum usable (net) speed is only half this value due to the protocol overhead and bus scheduling effort - please have a look at [this discussion](https://github.com/OpenHantek/OpenHantek6022/issues/60#issuecomment-591633973).

commit cadcafe8 (2020-11-30): calculate THD (optional), fix #132; disable spectrum for XY mode