OpenHantek/OpenHantek6022 #134: PR for DSO-2090 device

jedi7, 2020-12-03:
Hi,
If I create a PR for the DSO-2090 device, will ti be accepted? Or you want keep strictly only this 6022 device?

Ho-Ro, 2020-12-03:
It depends how complex the changes are. Please give me a rough estimation.
Long time ago [I removed support for  Hantek 2090, 2150, 2250 and 5200](https://github.com/OpenHantek/OpenHantek6022/tree/last_legacy_support) from the project, because I do not own any of these devices and cannot test during development. In the past I developed firmware and implementation for scopes I do not have access to, but these parts (DSO6022BL, DDS120) were very similar to the 6022BE and nevertheless it took some loops until success.

Ho-Ro, 2020-12-10:
> Please give me a rough estimation.

Ok - after 7 days it looks if it is not so important - I close here. Feel free to reopen it if you can provide more info.

jedi7, 2020-12-10:
Sorry I missed the previous response :( I had no time to look at it deeper. I just wanted to know if it will be ok for you, or you want keep strictly actual devices.

jedi7, 2020-12-10:
no need to reopen :) it is not issue. Just there was no other way to contact you :)

Ho-Ro, 2020-12-10:
As I wrote previously I'm not able to support *"virtual"* devices, i.e. a device I have not in my hands. But if the impact on the existing files is not too big we can try, otherwise you should provide your own branch with e.g. synced snapshots for the main releases.

closed 2020-12-27 by Ho-Ro

sl-maka, 2022-09-13:
How do I access this code change? I would very much like 2090 support but my own attempts have failed.

jedi7, 2022-09-14:
There was no time to write it :( and probably will not be for longer time.

Ho-Ro, 2022-09-14:
>  I would very much like 2090 support but my own attempts have failed.
Did you check how it was done in [openhantek](https://github.com/OpenHantek/openhantek)? This legacy project supports [DSO2090](https://github.com/OpenHantek/openhantek/blob/master/openhantek/src/hantekdso/models/modelDSO2090.cpp).

Ho-Ro, 2022-09-14:
- firmware ok?
- udev rule in place?

You could try to start with `--verbose 7` - this gives the most info, and then you can add more debug output in e.g. `finddevices.cpp`. I used a lot of `printf()` or `qDebug() <<` during development.

sl-maka, 2022-09-14:
Firmware hex is the one used by the legacy openhantek (although I dont know what to do with the *-loader.hex). Udev rule is already in place since im running legacy OH. But I tried replacing the udev rules with various those from github (OH and old OH6022). No diff. I dont think the firmware uploads. The "failed message" is immediate, no delay after choosing device (dso-2090 is only device in list).

I will test the verbose argument, thanks for the tip!

Ho-Ro, 2022-09-15:
>  I dont know what to do with the *-loader.hex). 

Have a look at 
[`UploadFirmware::startUpload(USBDevice *device)`](https://github.com/OpenHantek/OpenHantek6022/blob/ed658a0c19dd87cccf297a3abdb78589f1e439d0/openhantek/src/usb/uploadFirmware.cpp#L18) and compare it with the current code. IIRC all FW versions that wer extracted from a windows .exe had a `loader` and a `firmware` part, while my improved FW comes as one block. Just sprenkle a lot of `printf` in the legacy code and watch the steps.

sl-maka, 2022-09-16:
Can you simply skip the loader hex? Or are the firmware hexes used by OH6002 not the same as those used by OH? It seems the firmware hex is uploaded in my case. I can follow every SREC line in the debug printout. Then a reset is issued, then nothing. As if the firmware content was bad somehow.

You say "my improved FW". Is OH6022 using custom FW for its supported devices? And this in practice includes the functionality the loader supplied?

I will examine my logs, thanks!

Ho-Ro, 2022-09-16:
I can imagine that if there's a two-stage loading (*loader.hex & *firmware.hex) both should be loaded, compare [old](https://github.com/OpenHantek/OpenHantek6022/blob/ed658a0c19dd87cccf297a3abdb78589f1e439d0/openhantek/src/usb/uploadFirmware.cpp#L18) and [new](https://github.com/OpenHantek/OpenHantek6022/blob/b0da1cd8cd425cc2c2c947ceb344c23c3d103d06/openhantek/src/usb/uploadFirmware.cpp#L22).
I can't remember what actions the loader performed - maybe it initialises something in the EzUSB chip. OH6022 uses a completely new [free open source FW](https://github.com/Ho-Ro/Hantek6022API/tree/main/PyHT6022/Firmware) that does not depend on the Hantek code - it is based on the Sigrok project with extensions, while the 2090 uses the Hantek FW extracted from the Windows binary (as did the old OH and the last legacy version).
You don't have your development in GitHub - do you? If you set up [your own GitHub repo](https://docs.github.com/en/get-started/importing-your-projects-to-github/importing-source-code-to-github/adding-locally-hosted-code-to-github#adding-a-local-repository-to-github-using-git) from your local development it would be easier for us to communicate and for me to check some ideas - e.g. what does your ModelDSO2090::ModelDSO2090() look like, it might break the program if the FW version is wrong, with the same error screen you showed above. Also interesting would be the `dmesg` lines when you connect the scope and start the program.