OpenHantek/OpenHantek6022 #137: GUI suggestion

zen67, 2020-12-10:
Everything is grouped in one place, easy to tap without much mouse movement.
Everything happens faster.
A window space opens.
Is it possible to be a permanent / switchable option without dragging the tool bar?
Is there a possibility for 3D buttons?
[image]

Ho-Ro, 2020-12-12:
> Is it possible to be a permanent / switchable option without dragging the tool bar?

You can drag the tool bar to the right border once and it will stay there - also after restart.
[image]

> Is there a possibility for 3D buttons?
￼
I do not know - and honestly speaking I don't care about the buttons, the screen is important. For me the gui looks ok.

closed 2020-12-12 by Ho-Ro

zen67, 2020-12-12:
Linux build is OK.
Windows build is differеnt.
Еverything is smaller and 2D, unlike the Linux build, where the buttons are like 3d and everything is bigger and more convenient to work with.

Ho-Ro, 2020-12-12:
This is how it looks on a W10 computer (screen size 1920x1080) - for me it doesn't look unusable:
[image]

zen67, 2020-12-12:
I commented on the differences.
See the difference in the attached photos.
[image]
[image]

Ho-Ro, 2020-12-14:
Ok, I rename this issue with the clear target of a Windows GUI enhancement.

@zen67 wrote:
> Linux build is OK.
Windows build is differеnt.
Еverything is smaller and 2D, unlike the Linux build, where the buttons are like 3d and everything is bigger and more convenient to work with.

The GUI style was set for Linux by `openHantekApplication.setStyle( QStyleFactory::create( "Fusion" ) );` in source file [main.cpp](https://github.com/OpenHantek/OpenHantek6022/blob/d16b3dc9dd452d57b1dc9052469ba48f57556e17/openhantek/src/main.cpp#L142). I do not use any Windows computer for development and the trial and error process via the web based CI tool appveyor is slow, so I here is one last test that applies the same style also for windows (d16b3dc), also available  from *"[unstable](https://github.com/OpenHantek/OpenHantek6022/releases/tag/unstable)"*. 
This is how this looks fullscreen on a 1920x1080 win laptop:

[image]

zen67, 2020-12-14:
Thanks!
It's much better now, but it's still a little smaller than Linux.
Is it possible to have an autodetect of screen resolution to eliminate this forever?
That's what it looks 
[image]

Ho-Ro, 2020-12-14:
You can also play with the environment variable QT_SCALE_FACTOR

    set QT_SCALE_FACTOR=1.5
    OpenHantek.exe

[image]

zen67, 2020-12-15:
Thanks!
Font change works just fine!!!
Where in the source, can the font size be specified?
The font can also be changed through shortcut ("C: \ Users \ ..... \ Desktop \ Hantek \ OpenHantek-Win-x64 \ OpenHantek.exe" -f 10), but it is inconvenient to does every time (for example when you make a new shortcut)
Screenshot (font = 10)
[image]

Ho-Ro, 2020-12-15:
[main.cpp#L100](https://github.com/OpenHantek/OpenHantek6022/blob/5ef84a9524b0ef8c679750cb9ed7625a13b46e97/openhantek/src/main.cpp#L100): `const int defaultFontSize = 10;`

zen67, 2020-12-15:
const int defaultFontSize = 10;
increases font too much (screenshot 1)
When OpenHantek.exe -f 10 is used, the font becomes smaller and everything comes close to the Linux build
(screenshot 2).
(Screenshot 2 is from the last build, but OpenHantek.exe -f 10 was used.).
Screen 1
[image]

Screen 2
[image]

Ho-Ro, 2020-12-15:
So, next step - the font size can also be changed in the settings menu (and become active after a restart). Available in unstable:
[image]
Are you already able to build your own version for Widows as well as Linux?

zen67, 2020-12-15:
Great!
Changing the font works fine now.

_Are you already able to build your own version for Widows as well as Linux?_
Own - hardly.
I manage to create a 48MS/s version for the experiments I do with the hardware (through appveyor.com).