OpenHantek/OpenHantek6022 #138: Connection failed with 6022BL on Win10 20H2, used zadig, read the docs and comments...

dyarkovoy, 2020-12-19:
Using the latest openhantek 3.1.4 (also tried b483, b479 and b461, x64 and x86 variants).
I followed zadig way, tried all combinations of drivers (WinUsb, libusb-win32, libusbK), rebooted mutiple times, tried keeping the USB plug across installations/reboots, then tried re-plugging. Still I get the error "Connection failed! Couldn't open device: Operation not supported or unimplemented on this platform", but there's a twist: when I select "DSO-6022BL" from the drop-down in openhantek initial window (btw, it shows green), it immediately says "The device is ready for use", and this text stays up for less than a second, then the drop-down automatically changes to "DEMO" and the "Connection failed!" error appears. Now, if I'm fast enough to click OK during these milliseconds when "The device is ready for use" shows, the main window opens and it's almost functional, except that there's no graph, and I see "ERR" messages in place of the graph

zen67, 2020-12-19:
Can you remove any devices related to Hаntek (check in device manager) and install the driver in the archive?
Note that you need to check the "view -> show hidden devices" option. Hantek/DSO is located in: Universal serial bus devices
There are two files in the archive OpenHantek-....
Hantek_6022B.inf
Hantek_6022B.cat
Right button -> install -> Hantek_6022B.inf

Write here if there is still a problem.

dyarkovoy, 2020-12-20:
Nope, didn't work unfortunately.
Here's what I did:
- removed all devices related to Hantek in device manager (with "show hidden devices") under Universal serial bus devices
- installed driver from the OpenHantek archive
- rebooted
- verified that the right driver is being used for Hantek devices (it has Korean signature, hope it is the right one - see screenshot?)
- started OpenHantek, and seeing the same behavior - the device is available in the list, shows green, when I select it - shows "Ready for use" for under a second and then switches to DEMO with the error message.

Here are the screenshots

[image]

[image]

DaPa, 2020-12-20:
As said in [windows-usb-access](https://github.com/OpenHantek/OpenHantek6022#windows-usb-access) you must **not** use the original driver! Go do device manager and select Uninstall Device and ensure you tick "Delete the driver software for this device". Then follow the steps to install the Zadig driver **twice**!

Ho-Ro, 2020-12-20:
The screen shot shows measured values at the bottom of the scope (this means the USB communication is up and running). Try connecting the CH1 probe to the calibration out connector -> should give 2 Vpp and 1 kHz in the yellow bottom line. Your problem looks like you have issues with your OpenGL implementation of Windows, you could try the options `--useGLSL120` or `--useGLES`:
````
Usage: ./OpenHantek [options]

Options:
  -h, --help                 Displays this help.
  -v, --version              Displays version information.
  -d, --demoMode             Demo mode without scope HW
  -e, --useGLES              Use OpenGL ES instead of OpenGL
  --useGLSL120               Force OpenGL SL version 1.20
  --useGLSL150               Force OpenGL SL version 1.50
  -i, --international        Show the international interface, do not translate
  -f, --fontsize <fontsize>  Font size (default = 10, 0: automatic from dpi)
````

dyarkovoy, 2020-12-20:
Hi @DaPa, thank you for the input and sorry for the confusion. I did **not** use the original driver. If you take a look at my initial message in this issue, I **did** follow the instructions for the Zadig driver, did it twice! I followed your instructions in https://github.com/OpenHantek/OpenHantek6022/blob/master/docs/OpenHantek6022_zadig_Win10.pdf
carefully, and only after they didn't work for me, I submitted this issue.
Next, I was advised by @zen67 to install the driver that comes along with the OpenHantek (not the original driver from the Hantek manufacturer, I never installed it actually). So I did that, and was still having the described issue. 
I will again outline my full set of steps to avoid confusion.

1. My setup:
- Windows 10 20H2 build 19042.685
- working Hantek 6022BL (verified that it was working on **a different machine** 2 years ago when I bought it, with the software supplied by the aliexpress seller)
- No previously installed USB driver for Hantek on this computer

2.  Set of steps to install Zadig driver
- Downloaded the latest Zadig 2.5, started it, selected "List all devices"
- Selected Unknown device #1, installed WinUSB driver
- Rebooted, started OpenHantek, it updated the firmware and the new device appeared.
- Installed WinUSB driver for the new device with Zadig
- Rebooted, launched Zadig, verified that WinUSB driver is installed for DSO 6022BL device
- Launched Openhantek, got the error "Connection failed! Couldn't open device: Operation not supported or unimplemented on this platform" with the twist described in my first issue
- Repeated these set of steps for libusb-win32, libusbK - the problem was still persisting.

3. Set of steps suggested by @zen67 in his comment
- removed all devices related to Hantek in device manager (with "show hidden devices") under Universal serial bus devices
- installed driver from the OpenHantek archive
- rebooted
- verified that the right driver is being used for Hantek devices (it has Korean signature, hope it is the right one - see screenshot?)
- started OpenHantek, and seeing the same behavior - the device is available in the list, shows green, when I select it - shows "Ready for use" for under a second and then switches to DEMO with the error message.

4. Set of steps that I performed **again** following the @DaPa 's instructions here: https://github.com/OpenHantek/OpenHantek6022/blob/master/docs/OpenHantek6022_zadig_Win10.pdf (thinking I might have missed something)
- went to device manager, checked "show hidden devices", selected Uninstall Device and ensured I ticked "Delete the driver software for this device" for 2 devices I had under Universal Serial Bus devices (DSO-6022BL and Hantek 6022BL)
- rebooted
- verified there's no attached driver, the device now appears as Other devices -> Unknown device
- started OpenHantek and it predictably showed the error message, all good so far
- started zadig 2.5, it picked up the Unknown Device #1 with the ID 04B4:602A, clicked Install driver and waited for the success message.
- verified that I now have Unknown Device #1 under Universal Serial Bus devices in device manager, and the driver is correct (libwdi 6.1.7600.16385), exactly as shown on the picture in the pdf in step 4.
- restarted openhantek, it changed the firmware, and device ID changed to  04B5:602A, the device DSO-6022BL now appeared under Universal Serial Bus devices in device manager
- started Zadig again! selected  DSO-6022BL in Zadig and noticed that it already had WinUsb driver installed (this was the only difference with the pdf instructions). Verified that the driver is correct for DSO-6022BL device from Device manager, and yes - the driver is libwdi 6.1.7600.16385.  
- Opened openhantek.exe, when selecting now green DSO-6022BL, it says "The device is ready for use", and after 500-700ms reverts to DEMO and the error  "Connection failed! Couldn't open device: Operation not supported or unimplemented on this platform" appears.
- Just in case, to make sure I followed the instructions to a T, clicked Reinstall driver for DSO-6022BL in Zadig. Waited for success. Started openhantek.exe - no change. Rebooted. Started openhantek.exe - no change.

I'm thinking maybe this is some kind of unsupported flavor of 6022BL? I noticed device ID is  04B5:602A, not  04B5:6022 as in pdf

Ho-Ro, 2020-12-20:
> I'm thinking maybe this is some kind of unsupported flavor of 6022BL? I noticed device ID is 04B5:602A, not 04B5:6022 as in pdf

04B5:6022 -> 6022B**E**
04B5:602A -> 6022B**L**

dyarkovoy, 2020-12-20:
Hi @Ho-Ro, just tried the switches, you were right, `--useGLES` helps to remove errors and there's a graph now, thank you!
But still on the device selection page it behaves the same way - reverts to DEMO and errors after I select DSO-6022BL, shouldn't be related to the OpenGL, should it? I will see if I can fix my OpenGL.

Ho-Ro, 2020-12-20:
> I did not use the original driver ...

Where does the `DSO-6022BL` driver come?

[image]

If you install via [`Hantek_6022B.inf`](https://github.com/OpenHantek/OpenHantek6022/blob/master/cmake/Hantek_6022B.inf) file there should be only `Hantek 6022BL`:
````[Device.NTamd64]
"Hantek 6022BE" = Install, USB\VID_04B4&PID_6022
"Hantek 6022BE" = Install, USB\VID_04B5&PID_6022
"Hantek 6022BL" = Install, USB\VID_04B4&PID_602A
"Hantek 6022BL" = Install, USB\VID_04B5&PID_602A
````

dyarkovoy, 2020-12-21:
@Ho-Ro, I am not sure where the name DSO-6022BL comes from, but when I installed via Hantek_6022B.inf, both devices (DSO-6022BL and Hantek 6022BL) show exactly the same driver details (date,version,digital signer) as on screenshot. My guess is maybe the name is pulled from device itself? In any case, it should not matter if the driver is the same, should it?

dyarkovoy, 2020-12-21:
I guess we can say this issue is **almost** solved. Almost because I still have to quickly press "OK" on the device selection screen after selecting DSO-6022BL and before the error appears, to enter main window.
It looks like I am not able to solve OpenGL issue, because this is 7 years old laptop with Intel Mobile Series 4 video card that has no recent drivers with OpenGL.
If anyone runs into similar issue, the solution is:
1. follow https://github.com/OpenHantek/OpenHantek6022/blob/master/docs/OpenHantek6022_zadig_Win10.pdf 
2. run OpenHantek.exe with `--useGLES` switch
3. when you see the "Connection failed!" error, select your Hantek device from the drop down list, and immediately click "OK" before it changes back to DEMO with an error message.

closed 2020-12-21 by dyarkovoy

Ho-Ro, 2020-12-21:
Strange, I never saw this behaviour under Linux. What do you mean with 
> ... select your Hantek device from the drop down list ...

Are there more devices visible in this list (the list on the top of the window)?
[image]

Some thoughts:
* If you have any other devices connected to the USB busses then please remove them. 
* Is this an USB3 connector on your laptop?
* Connect the scope to a different USB plug.
* Use a different USB cable.
* Do you connect via an USB hub?

dyarkovoy, 2020-12-21:
>Are there more devices visible in this list (the list on the top of the window)?

I have 2 items in this list, the first is DEMO and the second is DSO-6022BL. I meant that I have to manually select  DSO-6022BL and immediately click OK, otherwise the selection automatically changes to DEMO with the error message, OK button becomes disabled.

I don't have any other devices connected to USB on this computer
There are 3 USB connectors on the laptop, none of them are USB3. I tried all 3, no diffierence
I'm not using a USB hub.
The only thing I didn't try is replacing the cable, the computer is at the remote location, I will try to do it once I'm there

dyarkovoy, 2020-12-25:
Update: tried replacing the cable, made no difference

Ho-Ro, 2020-12-27:
I put all your infos together and checked them against the data flow during device detection - one idea that came to my mind could be that your win installation reports sometimes bogus vendor ids (VID=0) that were mistaken as the DEMO device. Please check if the latest [windows artifacts](https://ci.appveyor.com/project/Ho-Ro/openhantek6022/build/job/09nri7ksttdo2sln/artifacts) solve your problem.

dyarkovoy, 2020-12-30:
Yes! This was it. I tried b492 and the issue is solved. Thank you.

Ho-Ro, 2020-12-30:
I have finally not only caught the symptom of bogus VIDs but [now](https://github.com/OpenHantek/OpenHantek6022/commit/40399244f829956d0127bf34ca58b08758a6cf5c) correctly ignore the demo device when detecting the devices available on USB. This version (without new functionality as b492) is available in [unstable](https://github.com/OpenHantek/OpenHantek6022/releases/tag/unstable). Thanks for your help to make OH6022 more robust.