OpenHantek/OpenHantek6022 #139: Unknown Error 2: (random errors). Also occurs in demo mode.

AtharvaMRFire, 2020-12-20:
Hi.
I installed the software on a new PC and when I start the program I get the following error:

[image]

This occurs on the demo mode too.

I moved to an new laptop. The older PC didn't show this error.
This laptop doesn't have the graphics display driver installed and I cannot install it, due to some reason and I am using the Microsoft Basic Display Driver(maybe related I don't know if the display driver is required).

Seemingly the issue is with my PC, I was wondering if you guys could point out what gives this error so I can fix it.
Just to be sure I also installed the latest version and it showed the same error.

AtharvaMRFire, 2020-12-20:
I went through a few older issues and I ran open hantek from cmd and used "start OpenHantek --useGLES"
Is it possible to start it in --useGLES from a shortcut?

AtharvaMRFire, 2020-12-20:
I did some research and found a way to create a shortcut to open using --useGLES
Step 1: Go to your desktop (or any other place you want the shortcut to be)
Step 2: Right click->new -> shortcut
Step 3: enter this in the target / location of the item:

"C:\Windows\System32\cmd.exe /c cd C:\Program Files\OpenHantek6022B & start openhantek  --useGLES"
Replace C:\Windows\System32\cmd.exe with the location of your cmd
Replace C:\Program Files\OpenHantek6022B with location of the folder which contains the program
Step 4:Next-> Enter name
You can also change the icon of the executable if you want.

AtharvaMRFire, 2020-12-20:
[image]

Now it seems to be working, but a part of the graph seems to get clipped off/shows error. Any solution to this?

AtharvaMRFire, 2020-12-20:
For the previous problem, turning off Histogram(left taskbar the one with bar graphs) solves the problems.

Devs, please see if anything can be done.

Ho-Ro, 2020-12-20:
> Now it seems to be working, but a part of the graph on the right side seems to get clipped off/shows error. Any solution to this?
For the previous problem, turning off Histogram(left taskbar the one with bar graphs) solves the problems.
Devs, please see if anything can be done.

It is clear that turning off the histogram doesn't show the histogram anymore - do you know what a [histogram](https://en.wikipedia.org/wiki/Histogram) is?
See e.g. this screenshot with the histograms of the square wave and the sine wave on the right: 
[image]

I'm glad that the issue is solved -> close.

closed 2020-12-20 by Ho-Ro

AtharvaMRFire, 2020-12-21:
Hi Dev, do you know what causes the first problem?

Ho-Ro, 2020-12-21:
The problem with the "unreadable" screen is caused by a not up-to-date or not fully compatible *OpenGL* driver on your system. Normally OH6022 uses *OpenGL* but you can force the (less complex) *OpenGL ES* (i.e. OpenGL for Embedded Systems) with the option `--useGLES`.

AtharvaMRFire, 2020-12-21:
Oh. So the problem was related to the graphics card driver. Thank you for your help, Martin.
We are all grateful for the efforts that you put in to maintain this FOSS.