OpenHantek/OpenHantek6022 #142: [Feature request] Please provide aarch64 deb

iamdavidcz, 2020-12-26:
Hi @Ho-Ro

It would great to have aarch64 .deb file as well since now Raspberry Pi OS supports 64-bit :)

Thanks!

Ho-Ro, 2020-12-27:
Hi, I build the arm binaries on one of my local RPi systems (while the Linux/Windows/MacOS binaries are built by CI provider appveyor on their machines). As I do not have any 64 bit RPi I provide only the 32 bit binaries. But if you have a native 64 bit RPi system and follow the [Linux build instructions](https://github.com/OpenHantek/OpenHantek6022/blob/master/docs/build.md#linux) it would not be too difficult to roll your own version. You can use the provided scripts [`LinuxSetup_AsRoot`](https://github.com/OpenHantek/OpenHantek6022/raw/master/LinuxSetup_AsRoot) and [`LinuxBuild`](https://github.com/OpenHantek/OpenHantek6022/raw/master/LinuxBuild).

closed 2020-12-27 by Ho-Ro

DoctorOgg, 2023-11-27:
Might be time to reopen this,

Ho-Ro, 2023-11-27:
Do you have a 64bit RasPi? Just follow my advice above.

DoctorOgg, 2023-11-27:
I do, however, it might be nice to see if your CI can now build those images. As im sure there are folks who not as capable as I am

Ho-Ro, 2023-11-27:
> if your CI can now build those images

My CI (GitHub Runner) creates Linux, MacOS and Windows packages. The RasPi and FreeBSD packages are created on my desk at home using a virtual FreeBSD and a real RasPi3 with a 32bit operating system. I do not have a RasPi with a 64bit operating system.
I only use Linux for all my development - so all other operating systems are a free gift to the community with no benefit to me. If you could provide a simple solution, I would be very grateful.

DoctorOgg, 2023-11-27:
I understood. Yeah, the latest version of rPi OS is 64bit; however, for Debian, at least, we don't need to build on a Raspberry Pi to get an arm version. GitHub Actions can build Debian packages for ARM, https://github.com/pi-top/debian-package-build-action