OpenHantek/OpenHantek6022 #143: udev rules not installed in correct directory

VictorLamoine, 2021-01-08:
**Describe the bug**
The udev rules are not installed in a directory parsed by `udev`.

**To Reproduce**
- Compile latest version (`git` repo)
- `make install`
- `udev` rules are installed in `/usr/local/lib/udev/rules.d/60-hantek.rules`

```bash
Install the project...
-- Install configuration: "RelWithDebInfo"
-- Up-to-date: /usr/local/lib/udev/rules.d/60-hantek.rules
-- Up-to-date: /usr/local/share/doc/openhantek/CHANGELOG
-- Up-to-date: /usr/local/share/doc/openhantek/LICENSE
-- Up-to-date: /usr/local/share/doc/openhantek/README
-- Up-to-date: /usr/local/share/doc/openhantek/HANTEK6022_AC_Modification.pdf
-- Up-to-date: /usr/local/share/doc/openhantek/OpenHantek6022_User_Manual.pdf
-- Up-to-date: /usr/local/share/doc/openhantek/OpenHantek6022_zadig_Win10.pdf
-- Up-to-date: /usr/local/share/applications/OpenHantek.desktop
-- Up-to-date: /usr/local/share/icons/hicolor/48x48/apps/OpenHantek.png
-- Up-to-date: /usr/local/share/icons/hicolor/scalable/apps/OpenHantek.svg
-- Up-to-date: /usr/local/bin/OpenHantek

```

Un-plug / plug the Hantek 6022, start the application and the applications errors about wrong permissions to access the device.

Quick fix, create a symlink:
```bash
sudo ln -s /usr/local/lib/udev/rules.d/60-hantek.rules /etc/udev/rules.d/
```

**Computer environment:**
 - OpenHantek version: 40399244f829956d0127bf34ca58b08758a6cf5c
 - OS: Kubuntu 18.04

Ho-Ro, 2021-01-08:
Do NOT install this way! Instead follow [the proposal](https://github.com/OpenHantek/OpenHantek6022/blob/master/docs/build.md#linux):
````
make -j4
rm -f packages/*
fakeroot make -j4 package
sudo apt install packages/openhantek_*_amd64.deb
````
Beside the possibility to remove cleanly this has also the benefit that dependency packages are installed automatically (they are marked also as automatically installed). If you want to remove OpenHanatek6022 and the dependencies are no longer needed you can remove them also (`apt autoremove`).

closed 2021-01-08 by Ho-Ro

VictorLamoine, 2021-01-08:
Got it thanks

You can cleanly uninstall with CMake:
```bash
sudo xargs rm < "install_manifest.txt"
```