OpenHantek/OpenHantek6022 #147: Not Responding when started in other than main display

ha-ves, 2021-01-12:
I began noticing this just recently. If you start the program in main display it'll work fine, but then you move it to other display and close it, next time it'll start from that display and it'll just White Screen and stopped responding, can't even move it.

**Steps to reproduce the behavior:**
1. Get more than single display/monitor
2. First time starting the program it starts on the main display and it's normal, everything is working fine
3. Now put the program to the other display, still normal
4. Close the program
5. Open the program, the start window still be in the main display
6. The working window is white-screen-ed on the other display

**Expected behavior**
Should work on any starting display.

**Computer environment (please complete the following information):**
 - OpenHantek version: *3.1.5*
 - OS: *Windows 10 ver 20H2 (64-bit)*
 - Video hardware: *GeForce Pascal GP104*
 - OpenGL version: *4.6*
 - *Pre-compiled Binary Win-x64-Release-b488*

**Scope device (please complete the following information):**
 - Device : *Hantek 6022BL*
 - Program top line *OpenHantek6022 (3.1.5) - Device DSO-6022BL (FW0207)*
 - other settings are default

**Additional context**
Multi-monitor setup

Ho-Ro, 2021-01-12:
Cannot reproduce your issue on a test system:
Microsoft Windows 10 Enterprise
10.0.17763 Build 17763
GL_VENDOR: Intel
GL_RENDERER: Intel(R) UHD Graphics 620
GL_VERSION: 4.6.0 - Build 26.20.100.7263

Ho-Ro, 2021-02-09:
closed after 4 weeks of inactivity

closed 2021-02-09 by Ho-Ro