OpenHantek/OpenHantek6022 #148: Automated CSV file saving or python compatibility for OpenHantek6022

mbobinger, 2021-01-12:
Thank you a lot for your great package! I've enjoyed using it on my RaspberryPi. I didn't manage to get PyVisa working with the RPI nor with a WINDOWS PC with the Hantek6022BE.

Since this is a low-cost PC oscope and widely used, it would be so practical to have it save CSV files automatially with predefined settings so the user can do some computation on the data.

Is there any way to automate data saving using OpenHantek?
Thank you!

Ho-Ro, 2021-01-13:
This issue was discussed more than once [earlier](https://github.com/OpenHantek/OpenHantek6022/issues/56#issuecomment-604696198), do you know the [python tools](https://github.com/Ho-Ro/Hantek6022API#use-the-device-as-a-data-logger) for storage and post-processing, e.g. [FFT](https://github.com/Ho-Ro/Hantek6022API/blob/master/examples/fft_from_capture.py).

Ho-Ro, 2021-01-26:
14 days without response - closed.

closed 2021-01-26 by Ho-Ro