OpenHantek/OpenHantek6022 #150: Horizontal marker

violagergo, 2021-01-14:
Hi!

I think horizontal markers would be very useful. 
Now I can measure only the time and frequency with the markers. For voltage measurement I need you use "Measure" menu which is not very easy to handle.
I ask you to create horizontal cursors the same way as we have the vertical, to see it on the main screen (like in the official Hantek software).

Thank you very much in advance!

Ho-Ro, 2021-01-14:
This was [already declined in 2019](https://github.com/OpenHantek/OpenHantek6022/issues/15#issuecomment-515941253), rationale:
> You have only one time scale for all voltage tracks, but with up to three voltage channels on the screen, you can get three different scales -> therefore the extra cursor section.

As a Q'n'D workaround to measure one voltage of the triggered channel you can abuse the trigger slider on the right, the trigger level is displayed in the top line at the left.

closed 2021-01-14 by Ho-Ro

violagergo, 2021-01-14:
Yes, but if we would have two markers, we could easily see the difference betweeen them, for example you could see the ripple of a signal.

violagergo, 2021-01-14:
[image]

In a case like this it is useful, when you want to see for example the "smaller" oscillation. You don't want to measure from the zero, and move the trigger and calculate, but you can see the result inmediatley.

Ho-Ro, 2021-01-15:
How would you solve this situation, the distance between red and green is 1/5 division, is it 10V for CH1 or 0.4V for CH2? What happens if you still have MATH and also three spectrum traces (volts and/or dB)?
All these problems are handled by the measurement add-on. 

[image]

P.S.: How do you work with a classical scope like my 4 channel monster - it is already a DSO but has neither analysis functions nor a text display?

[image]

violagergo, 2021-01-15:
I would solve it with a switch, where you can choose, which channel you want to measure. 

PS:
Your scope is beautiful , I love theese old fashion devices

Ho-Ro, 2021-01-15:
> I would solve it with a switch

All this is already available in the measurement window, you can even measure more than one channel, both time and amplitude. Since the vertical markers have a double purpose (measure and define the zoom range), they are accessible from the main window.