OpenHantek/OpenHantek6022 #151: RPI Gpio

stupidpersonthatisstupid, 2021-01-21:
Hi, im wondering is it possible to have like external knobs and switches for the software. I know RPI has Gpio-s which could be used with like switches and incremental encoders. Also I could emulate key presses with the arduino and just imput them over usb to like use shortcuts and change stuff. So my question is has anyone done this and is it possible? 

Thanks in advance :)

Ho-Ro, 2021-01-25:
Please provide a concept how to implement this in a generic way. I do not want to introduce regressions or additional support effort for the mainstream project caused by features that are only useful for a small user group.
Also I cannot support HW that is not on my desk.

spam-receiver, 2021-01-28:
It's not a problem. You can simply buy a conventional oscilloscope with controls, additionally equipped with a USB interface. But here software is being developed specifically for Hantek 6022BE/BL and its analogues.

spam-receiver, 2021-01-28:
> Hi, im wondering is it possible to have like external knobs and switches for the software. I know RPI has Gpio-s which could be used with like switches and incremental encoders. Also I could emulate [...]

It's not a problem. You can simply buy a conventional oscilloscope with controls, additionally equipped with a USB interface. But here only software is being developed specifically for Hantek 6022BE/BL and its analogues, and in the rare case, small hardware modifications to the oscilloscope itself.

spam-receiver, 2021-01-28:
> Please provide a concept how to implement this in a generic way. I do not want to introduce regressions or additional support effort for the mainstream project caused by features that are only useful for a small user group.
> Also I cannot support HW that is not on my desk.

Do we need an API for external control of OpenHantek6022? If you do it, you can bind anything to control the program. But is it really necessary?

vizkoze, 2021-01-28:
Music programs like ableton can use midi controllers to bind Rotary encoders and buttons to specific functions of the program.
Maybe midi is the way to go or can provide a example.

stupidpersonthatisstupid, 2021-01-28:
> 
> 
> > Please provide a concept how to implement this in a generic way. I do not want to introduce regressions or additional support effort for the mainstream project caused by features that are only useful for a small user group.
> > Also I cannot support HW that is not on my desk.
> 
> Do we need an API for external control of OpenHantek6022? Once created, anyone can bind anything to control the program. But is it really necessary?

Well, nothing in life in necessary except breathing, eating and drinking but it would be really nice to have a battery powered scope with like knobs and stuff bc I like having an oscilloscope with a big screen and touchscreen is not the best for scope. My plan was to use the 6022BE with a rpi to run the the software and then have an external arduino which would reed all the knobs and buttons and then simulate key presses over usb. Basically I would have a nice battery powered scope with a nice big screen and all the buttons and stuff.

BDW. sorry for a late reply to this thread, I was super busy this week :)

spam-receiver, 2021-01-28:
You can use this:

https://play.google.com/store/apps/details?id=com.martinloren.hscope

closed 2021-03-12 by Ho-Ro