OpenHantek/OpenHantek6022 #152: Display bug in rolling mode

VictorLamoine, 2021-01-26:
**Describe the bug**
The rolling mode displays wrong / changing values when used with long periods of time.

**To Reproduce**
- Use the oscilloscope test signal or generate an external slow pulse signal (~ 50 Hz is fine for this purpose), I use a STM32F103C8T6 for this.
- Connect the probes to the oscilloscope and signal
- Trigger mode: rolling (pick the right channel)
- Adjust trigger value
- Increase windows time to 500 ms / 1 sec

On the first signal roll everything is fine and after that the signal starts "dancing" and this makes no sense.

**Expected behavior**
The displayed signal amplitude should stay the same.

**Video**
If your browser says the file is corrupt, download and play with VLC, they are fine

1 kHz oscilloscope test signal:
https://filebin.net/jgb818sriwjsa8vl/integrated_signal.mp4?t=lb8irg89

External 40 Hz signal:
https://filebin.net/n1y8fqmxcphkkh67/slow_signal.mp4?t=cv0l1hby

**Computer environment :**
 - OpenHantek version: 25baf50
 - OS: Linux
 - Distribution, version: Kubuntu 20.04
 - Video hardware: nvidia Quadro K1100M
 - OpenGL version: server glx version string: 1.4
 - Qt version: Qt 5.9.5
 - C++ compiler version: gcc version 9.3.0 (Ubuntu 9.3.0-11ubuntu0~18.04.1) 

**Scope device:**
 - Device: Hantek 6022BE
 - Program top line [e.g. *OpenHantek6022 (Build ...) - Device ... (FW ...)*]
 - Input signal: calibration out 1 kHz / 40 Hz pulse signal
 - Probe setting: probe X1

Ho-Ro, 2021-01-26:
> the signal starts "dancing" and this makes no sense

Yes, this makes sense!
If you sample the 1 kHz signal with 2 kS/s then you violate the Nyquist rate requirement (fsample > 2 * f). e.g. you sample a 1000 Hz digital signal (0 1 0 1 0 1...) with 2000 Hz then you get (in theory) also 0 1 0 1... samples. OpenHantek6022 uses oversampling to improve SNR and to allow low sampling rates. If you downsample e.g. 1:10 then you average over 10 consecutive values and get (in best case) finally four samples A = 0, B = 1, C = 0, D = 1, but as the sampling is not in sync with your signal you may also get in worst case a = 0.5, b = 0.5, c = 0.5, d = 0.5 as seen below:
````
AAAAAAAAAABBBBBBBBBBCCCCCCCCCCDDDDDDDDDD
__________----------__________----------

aaaaaaaaaabbbbbbbbbbccccccccccdddddddddd
-----__________----------__________-----
````
The same applies to your 40 Hz (0000000001000000000100...) signal when sampling with 1 kS/s, you can "hit" the 1-pulse -> 1 or "miss" it ->  2 pulses of 0.5 each. As soon as the signal "rolls" the phase between signal and sample pulse is shifted uncontrolled (due to HW limitations of the Hantek6022 the samples are not gapless) and this gives all kind of bad structures. The roll mode is not useful for signals whose temporal course your eye can no longer follow.

[image]

As a rule of thumb you should select a sampling frequency that is minimal 4 times of the frequency of the interesting signal so that you get at least two samples for each pulse. In your case: to get your 40 Hz signal (with narrow pulses of 1 ms width) correct you should sample at least every 500 µs (-> 2 kS/s) to "catch" the 1 ms pulse correctly.
If you use the zoom function you can see the effect of faster sampling:
  
Too slow (500 Hz samples with 1 kS/s):
[image]

Fast enough (500 Hz sampled with 1 kS/s):
[image]

P.S:
> Program top line : ?

This line that gives you all info about hardware, software and firmware at one glance:
[image]

closed 2021-01-26 by Ho-Ro

VictorLamoine, 2021-01-26:
Thank you for all the details 👍🏼  I'm still a beginner with oscilloscopes.