OpenHantek/OpenHantek6022 #154: Configured sample rate seems to mismatch the actually used sample rate on DDS-120

markusstephany, 2021-01-28:
When using a SainSmart DDS-120 with the software, time / frequency measurement values of square signals (cursor measurements as well as automatically displayed measurement results) show different results depending on the configured sample rate.  

To reproduce, feed the input of any of the DDS-120 channels (both channels enabled) with a square signal of an appropriate amplitude, start the acquisition (using normal trigger at any edge). When changing the sample rate, the measured timings/frequency values change. 

Some examples:

Signal frequency 10 kHz, measured frequency 10.0 kHz at 5 MS/s, 11.0 kHz at 10 MS/s
Signal frequency 100 kHz, measured frequency 100.0 kHz at 5 MS/s, 110.0 kHz at 10 MS/s

(The following measurements are a on the edge of my equipment at home, so the signal is not a square anymore; the values do just slightly change in this area when setting up a different sample rate)
Signal frequency 1 MHz, measured frequency 999.6 kHz at 24 MS/s, 1.046 MHz at 30 MS/s
Signal frequency 2 MHz, measured frequency 1.999 MHz at 24 MS/s, 2.000 MHz at 30 MS/s

I would expect that (at least in the range up to 100 kHz), the sample rate based time calculations would not differ that much (=10%).

 - OpenHantek6022, version 3.1.5 release from this website
 - OS: Windows 10 Home 64 Bit

 - Device DDS120 (FW0100) 

TIA

closed 2021-01-28 by Ho-Ro

Ho-Ro, 2021-01-28:
This is another reminder to me that I really should only support the hardware (DSO6022BE) and also only the operating system (Debian) that I use and can therefore test. This simple bug existed untested in the system [for 8 months](https://github.com/OpenHantek/OpenHantek6022/commit/4836749efebc0b010b389fb3274174b5ea021f31) until you reported it.
The issue was caused by differrent sample rate coding of sigrok FW for DDS120 compared to my FW for Hantek DSO6022. The sigrok FW uses id=11 to switch to 10 MS/s (and id=10 for 100 kS/s), while my FW for 6022 uses the coding: id < 100 means sample rate in MS/s (e.g. 10 -> 10 MS/s) and id = 1NN means NN0 kS/s (e.g. 102 -> 20 kS/s).

markusstephany, 2021-01-28:
Thanks a lot for providing a fix! As soon as I get some free time, I will repeat the measurements and see what happens ;-)  11 to 10 is indeed a nasty conversion...

commit 13b2f5f0 (2021-01-28): fix #154 caused by different sample rate coding of sigrok DDS120 FW