OpenHantek/OpenHantek6022 #158: udev rules don't seem to be working for 6022BL

Fl0wless, 2021-02-08:
Just to let you know, the udev rules don't seem to be working for the 6022BL on arch linux, hat to remove them for the software to find the scope. Unfortunately I don't know too much about udev myself so I couldn't really help with that ;(

Ho-Ro, 2021-02-08:
> hat to remove them

You're mentioning `/usr/lib/udev/rules.d/60-hantek.rules`? After removing this file the program worked flawlessly?

Maybe there's another rule floating around... Please provide the output of

    grep -i 602a /etc/udev/rules.d/*.rules /lib/udev/rules.d/*.rules

Ho-Ro, 2021-02-22:
Closed after two weeks without any input - _war wohl nicht wichtig_ :)

closed 2021-02-22 by Ho-Ro

Fl0wless, 2021-02-27:
> Closed after two weeks without any input - _war wohl nicht wichtig_ :)

Sry, hatte das Issue ein bissel vergessen. Btw muss wohl an mir gelegen haben, es existieren wieder rules... naja, trotzdem super Projekt!