OpenHantek/OpenHantek6022 #159: Save Settings modal dialog not completely visible when sampling is running

pedropedro, 2021-02-27:
In the newest openhantek_20210222-bf1bdc2_amd64.deb installed on newest Xubuntu 20.10 (debian), the main window works fine, settings dialog too. When trying to manually save the settings **when the oscilloscope is running**, the "Save settings" modal dialog will pop up, but there is only the dialog frame visible, window "content" is transparent, one can see thru the frame the main window with the signal being displayed and refreshed.

Last stable version, Demo and bf1bdc2 with **stopped** sampling, work fine.

Xubuntu runs in a VirtualBox 6.1.16, with Additions.

[image]

spam-receiver, 2021-03-01:
The file manager dialog opens in the background.
[image]

Ho-Ro, 2021-03-02:
Strange, under native Debian stable the file select dialog opens on top of the scope window as expected. When I click on the scope window its title bar changes to dark (selected), but the scope's ui keeps inactive and the file dialog stays in the foreground - also as expected.
[image]
A W10 installation in demo mode shows also expected behaviour.

spam-receiver, 2021-03-02:
> Strange, under native Debian stable the file select dialog opens on top of the scope window as expected. When I click on the scope window its title bar changes to dark (selected), but the scope's ui keeps inactive and the file dialog stays in the foreground - also as expected.

KDE?

I have GNOME 3.38.3 with GTK 3.24.26 in updated Fedora 33.

Ho-Ro, 2021-03-02:
Yes, it's KDE/Plasma 5.14.5.1-1 with Qt 5.11.3+dfsg1-1+deb10u4 in pure Debian stable. But almost same behaviour with LXQt/Openbox (the file dialog is centered on the screen with KDE, while Openbox centers it on top of the OH application).
I have just built a version that uses the Qt file dialog instead of the native systemwide one, it will hit unstable within minutes, stay tuned...

pedropedro, 2021-03-02:
In the Xubuntu is xfce4 4.14, gnome-3-34/8, Qt5-gtk-platformtheme 5.14.2+dfsg-6 installed.

spam-receiver, 2021-03-02:
In 530b7f7 problem seems solved.
[image]

pedropedro, 2021-03-02:
"Save settings" works in 530b7f7, I have just tested it. Okay, it is not the same dialog type ... BTW I personally find it better  ;-)

@spam-receiver : Again, we do NOT speak here about the "Open file" ! That works fine, also in the previous version.

spam-receiver, 2021-03-02:
But "Export CSV.." still hiding in background.

pedropedro, 2021-03-02:
Yes, "Export CSV" is initially rendered as a frame only, it is using the other ("old" - 3.1.5.) dialog style. But here is sufficient to minimise the main window and the "Save CSV" dialog is suddenly (!) rendered w/o problems, all controls are also enabled => one can save the csv file ...

spam-receiver, 2021-03-02:
[image]
[image]

spam-receiver, 2021-03-02:
I have compiled the code from source. And you?

spam-receiver, 2021-03-02:
@Ho-Ro, look at this:

https://bugs.launchpad.net/elementaryos/+bug/1220179

> Giovanni Tirloni (gtirloni) wrote on 2019-01-20:
> 
> Experiencing this with Fedora 29 and GNOME 3.30.2.
> 
> How to reproduce:
> 
> 1. Open Chrome browser or gnome-terminal
> 2. Hit 'Ctrl-N'
> 3. New window pops up behind the window with focus
> 
> When that works fine:
> 
> 1. Open Chrome browser
> 2. Hit 'Ctrl-Shift'N' (incognito)
> 3. Window opens in front of existing windows
> 
> This happened in the past with Fedora 28 and unexpected started to work fine. Now it's back.
> 
> Disabling all extensions doesn't seem to help.

I think, that problem not in your code. It is GNOME libraries bug.

spam-receiver, 2021-03-03:
I found that this is the only program except OpenHantek6022 that opens the file manager window in the background. Bug in Qt libraries?
[image]
[image]

Ho-Ro, 2021-03-03:
Wow @pedropedro you opened a can of worms :)
But nevertheless, I also like the new dialogues better. And as the scope is a technical tool I'm not so picky about undisturbed "user experience". Most time I have some terminal windows open with midnight commander as my favorite file manager - this also doesn't fit into some shiny UI rules.
Regarding CSV, I added the line `fileDialog.setOption( QFileDialog::DontUseNativeDialog );` in `exportcsv.cpp`.
Now it should also use the intrinsic Qt dialog instead of the system version. Soon available in unstable...

> @Ho-Ro I'm speaking about the File -> Save settings as... dialog, not File -> Open settings ! ... that dialog works fine

When you opened your issue, I also checked the "Save settings as..." dialogue as well as the "Screenshot...", "Hardcopy...", "Print screen..." and "Export CSV..." dialogues and it made no difference - they just worked :)

pedropedro, 2021-03-03:
Okay, I tested the csv dialog too, I propose to close the issue and to go fish ... when the worms are creeping yet around ;-)

closed 2021-03-03 by pedropedro