OpenHantek/OpenHantek6022 #161: RPi 4B Touch Screen

Peter847, 2021-03-09:
I have an issue running OpenHantek 3.2 on an RPi with a 7" touchscreen using the latest version of the default (Debian) OS.  I lose the bottom quarter of the screen so, for example I cannot see anything below the "Configuration" title on the settings page.  Although I do see the bottom of the actual scope display.  I suspect this may be more of a Pi issue than an OpenHantek issue as the touchscreen has a non-standard resolution of 800 x 480.  I have setup the Pi to recognize this resolution and it seems to work on ok on the standard desktop.

Any thoughts on what I could do to fix it would be much appreciated!

Ho-Ro, 2021-03-12:
You can shrink the main window down to 790x299 (on my KDE Debian system) when you tab the docks. But the config screen has a fixed size of 844x615 on my system due to the chosen software architecture. A quick workaround (you do not use this screen permanently) could be to use `<ALT><mousedrag>` to the lower part into reach.

[image]

closed 2021-03-12 by Ho-Ro

Peter847, 2021-03-13:
Martin

Many thanks for the information.  I used the screen sizes to set
framebuffers in the Pi and I can now see the whole page.  However I am
getting a little confused on how the panel/tabs work but I don't see
anything in the manual on them?

Peter

Ho-Ro, 2021-03-15:
The tabbed docks are standard Qt QDockWidgets, you can grab them by their title line (e.g. Voltage, Horizontal, ..), rearrange them or drag them around on your screen.

[image]

To put them back just click the top right small button or drag them over the docking area until an **empty** blue space opens and drop it there.

[image]

If you drag a floating dock over another dock the target dock will turn blue. Drop the dock and it will be tabbed together with the target dock. 

[image]

This way you can combine all docks into one tab as I did in my 1st screen shot some days ago. Also the order of the tabs can be changed by dragging them left or right.

I will update the manual next time I touch it.