OpenHantek/OpenHantek6022 #170: Any chance of getting a Flatpak package?

kobus-v-schoor, 2021-03-19:
**Is your feature request related to a problem? Please describe.**
I think having flatpak as an option for installation would be great. It will allow sandboxing the application and distributing updates directly to the users.

**Describe the solution you'd like**
The software just needs to be packaged using flatpak, instructions here: [Building your first Flatpak](https://docs.flatpak.org/en/latest/first-build.html)

**Describe alternatives you've considered**
Snap is a similar solution but I personally like Flatpak much better. I find it much less problematic to use, and I often times prefer using it for applications which aren't in the official Debian repos yet or that are out of date. For instance, I use the Arduino IDE Flatpak instead of the Debian package

**Additional context**
N/A

kobus-v-schoor, 2021-03-19:
I'm currently busy trying to see if I can get this working, I'll let you know if I can conjure up something useful

kobus-v-schoor, 2021-03-19:
I've gotten it to successfully build and run with the following manifest:

```yaml
app-id: org.openhantek.OpenHantek
runtime: org.kde.Platform
runtime-version: "5.15"
sdk: org.kde.Sdk
command: OpenHantek
finish-args:
  - --socket=wayland

modules:
  - name: libusb
    sources:
      - type: archive
        url: https://github.com/libusb/libusb/releases/download/v1.0.24/libusb-1.0.24.tar.bz2
        sha256: 7efd2685f7b327326dcfb85cee426d9b871fd70e22caa15bb68d595ce2a2b12a

  - name: openhantek
    buildsystem: cmake
    sources:
      - type: archive
        url: https://github.com/OpenHantek/OpenHantek6022/archive/refs/tags/3.2.tar.gz
        sha256: 68d8111a0241fa28f6e2b0bd415552d0a0ac496fa7ec2ccd4747f1bbbc8ad985
```

I'll probably be getting my scope tomorrow (I don't have it yet), after which I'll hopefully be able to test the USB access (which is almost certainly not currently working). I'll post any other updates here

kobus-v-schoor, 2021-03-20:
I'm going through the procedure to submit the package to Flathub now, but I think it would be better if you are the maintainer of the package. I believe I'll be able to transfer the repo to you should you want to be in charge of maintaining it, all that would be needed from your side is to update the URL and SHA256 sum under the "openhantek" module when you release updates.

Ho-Ro, 2021-03-20:
I prefer you to take care of the Flatpack. I do not use this package format and can therefore not guarantee the correct function.
This procedure also happens, for example, for the Fedora packages that @Vascom maintains.

kobus-v-schoor, 2021-03-21:
No problem, I'd be happy to maintain the package since I'll be using it anyway.

Ho-Ro, 2021-03-21:
Cool, if everything is up and running I'll put a link on the main `readme.md`.

kobus-v-schoor, 2021-03-22:
The app has been published to Flathub (https://flathub.org/apps/details/org.openhantek.OpenHantek6022) and is ready to go. Adding the udev rules to /etc does still needs to be done manually, so I've added a note in the app's description. I'll be updating the app's metadata somewhere in the near future so that it is more complete, but for now the basics are there. The manifest (build instructions) and metadata is being managed in the following repo: https://github.com/flathub/org.openhantek.OpenHantek6022. If you need me to change something it is easiest if you open an issue/pull-request there, and all Flatpak related issues should probably be opened there as well. I'm subscribed to this repo so I should get a notification when you make a new release, after which I'll publish the new release on Flathub in a timely fashion.

Users that have flatpak installed should now be able to install with just a `flatpak install flathub org.openhantek.OpenHantek6022` :)

I'd just like to say thank you for maintaining this wonderful piece of software, it is the only reason I even considered buying the Hantek scope :)

closed 2021-04-19 by Ho-Ro