OpenHantek/OpenHantek6022 #175: Windows 10 cyrillic username problem

rkurbatov, 2021-03-28:
**Describe the bug**
The program cannot upload firmware to the device. 

**To Reproduce**
I tried several drivers without success. As I'm using the Far Manager (good old Linux habit) I saw the warning message in console, it could not open some file in C:\Users\some-strange-symbols\Temp\OpenHantek.someId folder. 

When I managed to rename my user folder so it did not contain cyrillic letters the problem disappeared.

Not sure if you can fix it easily for Windows but it least it could help and save some time for other people with non latin letters usernames.

Ho-Ro, 2021-03-29:
Strange - OH uses a normal Qt function to create a temporary file from the firmware resource in ` QDir::tempPath()` (on Windows this is usually the path in the TEMP or TMP environment variable):
[`temp_firmware_path = std::unique_ptr< QTemporaryFile >( QTemporaryFile::createNativeFile( firmwareRes ) );`](https://github.com/OpenHantek/OpenHantek6022/blob/d7d6c2236bc49a2f52da0a9cd36d4a2d505c21e5/openhantek/src/usb/uploadFirmware.cpp#L37)
and converts this file name to `const char *path = temp_firmware_path->fileName().toUtf8().constData()` 
that is transferred to `fopen()`.
You could try to change `%TEMP%` or `%TMP%` to something else where you have write access, e.g. `c:\temp`.

@spam-receiver: Did you encounter this kind of problem

spam-receiver, 2021-03-29:
I confirm the bug. If the username is Cyrillic in Windows 10, then the oscilloscope is not visible. The program offers to start demo mode. I often faced such problems with various programs, so, just in case, I become to use the username in Latin, both in Windows and Linux.

Ho-Ro, 2021-03-30:
Hello, friends of Cyrillic symbols, please check [this test build](https://ci.appveyor.com/api/buildjobs/vs362i1542hjpy0g/artifacts/build_x64%2Fopenhantek%2FOpenHantek-Win-x64-Release-b539.zip).
If it does not help, you could try to set `%TEMP%` and `%TMP%` to e.g. `C:\Windows\Temp` instead of the default `C:\Users\xxx\AppData\Local\Temp`.

Vascom, 2021-03-30:
Linux users need not worry?

spam-receiver, 2021-03-30:
OpenHantek-Win-x64-Release-b539.zip works normally.

Ho-Ro, 2021-03-31:
> ... works normally.

"normally" - hopefully in the sense of "as expected" and not in the sense of "as before". :)
Thank you for your support, I will wait for @rkurbatov's "OK" before closing this issue.

Ho-Ro, 2021-04-01:
It is available in [unstable](https://github.com/OpenHantek/OpenHantek6022/releases/tag/unstable) for testing. If I do not receive failure reports I will close this issue after Easter.

rkurbatov, 2021-04-01:
Wow, that was really quick.

Sorry, I'm using Linux normally, but attached this box to my old spare Windows desktop (just in case, you know).
It works normally now. I checked on Administrator account (that is Администратор on my machine) - old version searches for device, b539 just runs normally. Thanks!

closed 2021-04-01 by rkurbatov

Ho-Ro, 2021-04-01:
Roman, thx for reporting this _sleeping_ issue. This project lives from user feedback.