OpenHantek/OpenHantek6022 #176: Firmware won´t upload

Schwarzi07, 2021-03-29:
I have a 6022BL and everytime I try to connect i get an error saying something along the lines of "Could´nt open device", I cant read he full error because it dissapears quickly. I have tried running it as Admin with no effect. I have also looked at #175 and made sure I don´t have Cyrillic characters in my Name.

Schwarzi07, 2021-03-29:
I´m using Windows 10 pro and have the scope connected directly connected to the motherboard. I have 8gb of ram and an intel celeron n3060 at 1,6 GHz.

Ho-Ro, 2021-03-29:
You should provide some more info if you really want help:

**Computer environment (please complete the following information):**
 - OpenHantek version: [use always the latest commit, e.g. *release v2.03, commit 1a2b3c4, ...*]
 - OS: [e.g. *linux, macOS, windows, ...*]
 - Distribution, version: [e.g. *debian stable amd64, ...*]

**If you compiled the program on your PC**
 - Qt version: [e.g. *Qt 5.11.x*]
 - C++ compiler version: [e.g. *gcc 8.3.x*]

**Additional context**
Add any other context about the problem here, e.g. the program output if started in a console like xterm.

As you wrote "Admin" I suppose you're on Windows. Did you install the correct USB driver? Try starting the program from cmd.exe and check for additional error messages.

Schwarzi07, 2021-03-29:
I have managed to capture a screenshot of the error by using an older version . 

[image]

At this point i have tried every version of OpenHantek with no real advancements. I have installed the latest driver from the Hantek website. I have also tried using PulseView from Sigrok with no avail.

Ho-Ro, 2021-03-30:
>  I have installed the latest driver from the Hantek website.

Did you read [the doc](https://github.com/OpenHantek/OpenHantek6022#windows-usb-access)? It says:
> **The original Hantek driver for Windows doesn't work!**

Either follow [this](https://github.com/OpenHantek/OpenHantek6022/blob/master/docs/build.md#microsoft-windows-usb-driver-install-with-zadig) or [that](https://github.com/OpenHantek/OpenHantek6022/blob/master/docs/build.md#signed-winusb-driver-for-hantek-6022bebl) approach!

Two remarks:
- Because I want to avoid this kind of *sliced* information I offer the issue template when opening an issue. It does make sense to provide as much info as possible upfront to easy **my** task of helping **you**.
- It was not by chance that I wrote: *No support for non-Linux related issues unless a volunteer steps in!* Most Linux installations work flawlessly, same goes for FreeBSD and RasPi (more or less Debian based). It may sound philosophical, but Linux users understand the Zen of the computer better. :)

closed 2021-03-30 by Schwarzi07

Ho-Ro, 2021-03-30:
Ok issue solved, congrats. Would you mind sharing your solution with us?

Schwarzi07, 2021-03-30:
0. Boot into safe mode 
1. Remove original driver 
2. Remove hantek software
3. Reboot pc in order to exit safe mode 
4. Install custom driver 
5. Reboot 
6. Run OpenHantek as Administrator 

If this dorsnt work, try creating an Linux vm and pass the oscilloscope through to the vm. 

Also make sure that the oscilloscope isn't connected via a USB hub. If you are still  having issues try using a different USB cable because the one included isn't proper shielded.