OpenHantek/OpenHantek6022 #184: [W10] OpenHantek6022 wont even show up, Qt5 related issue found in Windows logs

ChristopheMineau, 2021-04-22:
**Describe the bug**
On a fresh W10 laptop  install, installed the last W10 packages  (OpenHantek-Win-x64-Release-b542 , b488 as well) , installed the usb driver using both proposed methods.
When launching OpenHantek6022, the dark blue window just flickers and disappears.
A crash report is found in the Windows logs.

**To Reproduce**
**IMPORTANT:** Always unplug and replug the scope if used before with a different application, e.g. sigrok or the original Hantek software.
**No support for non-Linux related issues unless a volunteer steps in!**
(glups, I am ok to volunteer, but need some clue ..)

Steps to reproduce the behavior:
1. Go to '...'   well as sai above ...
2. Select '....'
3. See error

**Expected behavior**
it should work ...

**Screenshots**
If applicable, add screenshots or a video to help explain your problem.

**Computer environment (please complete the following information):**
 - OpenHantek version: OpenHantek-Win-x64-Release-b542
 - OS: windows 10
 - Distribution, version: [e.g. *debian stable amd64, ...*]
 - Video hardware: Intel HD Graphics
 - OpenGL version: [e.g. *OpenGL 4.5, Mesa 19.0.3*] dunno
 - Qt version: [e.g. *Qt 5.11.x*]  dunno
 - C++ compiler version: [e.g. *gcc 8.3.x*] dunno

**Scope device (please complete the following information):**
 - Device Hantek 6022BE
 - Program top line [e.g. *OpenHantek6022 (Build ...) - Device ... (FW ...)*]
 - Input signal: [e.g. *calibration out 1 kHz, sine wave 1 Veff @ 50 Hz, ...*]
 - Probe setting: [e.g. *no probe, probe X1, probe X10, ...*]

**Additional context**
Here is the log found reporting this crash :
Nom de l’application défaillante OpenHantek.exe, version : 0.0.0.0, horodatage : 0x607c891a
Nom du module défaillant : Qt5Gui.dll, version : 5.15.2.0, horodatage : 0x5fa4ddbb
Code d’exception : 0xc0000005
Décalage d’erreur : 0x0000000000071075
ID du processus défaillant : 0x2210
Heure de début de l’application défaillante : 0x01d7375526a345b5
Chemin d’accès de l’application défaillante : C:\Installations\openhantek_20210418-dcac3f7-1_win_x64\OpenHantek.exe
Chemin d’accès du module défaillant: C:\Installations\openhantek_20210418-dcac3f7-1_win_x64\Qt5Gui.dll
ID de rapport : bbfb1f0e-2883-481d-b15b-ceabb2d7715f
Nom complet du package défaillant : 
ID de l’application relative au package défaillant :

Ho-Ro, 2021-04-22:
**additional information:**

> last W10 packages (OpenHantek-Win-x64-Release-b542 , b488 as well)

- last is 3.2.2, built with VS2019, Qt 5.15.2.0
- b542 is 3.2.1, built with VS2019, Qt 5.15.2.0
- b488 is 3.1.5, built with VS2015, Qt 5.11.3.0

>Nom de l’application défaillante OpenHantek.exe, version : 0.0.0.0, horodatage : 0x607c891a
Nom du module défaillant : Qt5Gui.dll, version : 5.15.2.0, horodatage : 0x5fa4ddbb
Code d’exception : 0xc0000005
Décalage d’erreur : 0x0000000000071075
ID du processus défaillant : 0x2210
Heure de début de l’application défaillante : 0x01d7375526a345b5
Chemin d’accès de l’application défaillante : C:\Installations\openhantek_20210418-dcac3f7-1_win_x64\OpenHantek.exe
Chemin d’accès du module défaillant: C:\Installations\openhantek_20210418-dcac3f7-1_win_x64\Qt5Gui.dll
ID de rapport : bbfb1f0e-2883-481d-b15b-ceabb2d7715f
Nom complet du package défaillant :
ID de l’application relative au package défaillant :

Name of the failed application OpenHantek.exe, version: 0.0.0.0, timestamp: 0x607c891a
Name of the failed module: Qt5Gui.dll, version: 5.15.2.0, timestamp: 0x5fa4ddbb
Exception code: 0xc0000005
Error offset: 0x0000000000071075
ID of the failed process: 0x2210
Start time of failed application: 0x01d7375526a345b5
Path of the failed application: C:\Installations\openhantek_20210418-dcac3f7-1_win_x64\OpenHantek.exe
Path to failed module: C:\Installations\openhantek_20210418-dcac3f7-1_win_x64\Qt5Gui.dll
Report ID: bbfb1f0e-2883-481d-b15b-ceabb2d7715f
Full name of the failed package :
Application ID for the failed package:

Translated with www.DeepL.com/Translator (free version)

**1st guess**
You could execute the program from cmd.exe and see if it writes more error messages.
The three OH Win versions above wer downloaded some hundred times in the past and I never got a report about this type of error, so I suspect that it is caused by your specific system or the fact that you did a fresh W10 install (there is [another W10 crash](https://github.com/OpenHantek/OpenHantek6022/discussions/174#discussioncomment-642436), but it shows at least the device selection screen). 
Did you install the supplied `vcredist_x64.exe`? [support.microsoft.com: the-latest-supported-visual-c-downloads](https://support.microsoft.com/en-us/topic/the-latest-supported-visual-c-downloads-2647da03-1eea-4433-9aff-95f26a218cc0)

Ho-Ro, 2021-04-23:
You could try the latest [unstable](https://github.com/OpenHantek/OpenHantek6022/releases/tag/unstable) release and use the cmd line option `--verbose` that will show the detailed program steps during startup until the main window opens, this will not solve your issue but narrow it down (hopefully) to the crash.

ChristopheMineau, 2021-04-23:
Hi Ho-Ro and thank you for taking care of this case.
Sorry, for the translation, I should hav done it by myself.
So I did exactly what you suggested :

- Took the last unstable version openhantek_20210423-0415321-1_win_x64.zip
- I installed the sb driver from there using "install" on Hantek_6022B.inf
- I installed the vc_redist.x64.exe again.
- restarded the PC,
- replugged the Hantek on
- opened a powershell in the openhantek_20210423-0415321-1_win_x64 folder
- used the verbose option, here is the result :
- `PS C:\Installations\openhantek_20210423-0415321-1_win_x64> .\OpenHantek.exe --verbose
11 ms: OpenHantek6022 - version 20210423 - commit 892
23 ms: create openHantekApplication
238 ms: set "Fusion" style
243 ms: load translations for locale "fr_FR"
245 ms: init libusb
272 ms: show splash screen
7737 ms: use device "DSO-6022BE" serial number "811244AB11C9"
7738 ms: create DSO control thread
7757 ms: create settings object
7766 ms: create exporters
7766 ms: create post processing objects
7767 ms: setup OpenGL
Attribute Qt::AA_ShareOpenGLContexts must be set before QCoreApplication is created.
7791 ms: set QFont(Arial,10,-1,5,50,0,0,0,0,0)
7808 ms: create main window
QWindowsEGLContext: ANGLE only partially supports OpenGL ES > 3.0
QWindowsEGLContext: Failed to create context, eglError: 3005, this: 0x17cfc8c4ce0
QOpenGLFunctions created with non-current context
PS C:\Installations\openhantek_20210423-0415321-1_win_x64>`

I guess it's an issue with my installation of OpenGl, but how could I handle that?
Thanks again for your support.
C.M.

Ho-Ro, 2021-04-24:
> Sorry, for the translation, I should have done it by myself.

Pas de problème, je peux lire le Français ;)

From your log above (I edited it to make it more readable) I see that the program starts, opens the splash/selection screen and finally crashes when it comes to OpenGL. This was not clear for me from your 1st post, I suspected an immediate crash. So what you could do is similar to [my hint](https://github.com/OpenHantek/OpenHantek6022/discussions/174#discussioncomment-648245) for @DVEHUB72 in the discussion area.
Dunno if the message `Attribute Qt::AA_ShareOpenGLContexts must be set before QCoreApplication is created.` is relevant, my Debian doesn't complain. Nevertheless I will change this just for testing purposes in the next commit.

Ho-Ro, 2021-04-24:
`OpenHantek.exe --useGLES` solved the issue for @DVEHUB72:  https://github.com/OpenHantek/OpenHantek6022/discussions/174#discussioncomment-653831

ChristopheMineau, 2021-04-26:
Hi, 
Great Martin, I confirm that` OpenHantek.exe --useGLES` works just fin !

ChristopheMineau, 2021-04-26:
Martin, 
I have just tried you test commit , still there is an open GL problem with it.
The oscillo Gui now opens, but with this error message griht in the middle :
[image]

Meanwhile, with the  --useGLES option, it works fine.
Thanks for your support !

closed 2021-04-26 by Ho-Ro

Ho-Ro, 2021-04-26:
I took your and @DVEHUB72's info as an opportunity to do the long-planned refactoring of [`glscope.cpp`](https://github.com/OpenHantek/OpenHantek6022/blob/main/openhantek/src/glscope.cpp).

Some graphic systems promise a higher version of OpenGL, although certain SL (shading language) functions are not fulfilled. This apparently occurs sometimes under Windows. In the event of an error, I now automatically switch to the syntax of the older stable OpenGL SL version 120. If this does not help, I at least display a hint for the user to switch to the simpler variant OpenGL ES (OpenGL for embedded systems) with --useGLES; unfortunately, this selection can no longer be made automatically at this late stage.

Please check version [openhantek_win_x64_44](https://github.com/OpenHantek/OpenHantek6022/suites/2586871764/artifacts/56465952).

ChristopheMineau, 2021-04-27:
Hi Martin,
Unfortunately, still an error for me with this 44 package :
[image]

But the fallback solution is good enough for me, you can close this case.
Thanks again for the good work, this version of the DSO is so much better than others :)
Christophe

Ho-Ro, 2021-04-27:
Hi Christophe, your OpenGL installation does not support SL (shading language) version 150 or 120, so you have to select (by hand) OpenGLES - as the error message tells you (there is just the newline missing). Please show the error message when you start OpenHantek from the command line (without --useGLES) because this shows the installed OpenGL version.

ChristopheMineau, 2021-04-27:
> Hi Christophe, your OpenGL installation does not support SL (shading language) version 150 or 120, so you have to select (by hand) OpenGLES - as the error message tells you (there is just the [...]

Here is the error log 

 ```
PS C:\Installations\openhantek_20210426-62c54e6-1_win_x64> .\OpenHantek.exe
Attribute Qt::AA_ShareOpenGLContexts must be set before QCoreApplication is created.
Attribute Qt::AA_UseOpenGLES must be set before QCoreApplication is created.
QOpenGLShader::compile(Vertex): ERROR: 0:2: '150' : version number not supported
ERROR: 0:2: 'in' : storage qualifier supported in GLSL ES 3.00 and above only

*** Problematic Vertex shader source code ***

          #version 150
#line 2
          in highp vec3 vertex;
          uniform mat4 matrix;
          void main()
          {
              gl_Position = matrix * vec4(vertex, 1.0);
              gl_PointSize = 1.0;
          }

***

OpenGL version "OpenGL ES 2.0 (ANGLE 2.1.0.57ea533f79a7)"
To get rid of this error message, try the command line option '--useGLSL120' or '--useGLES'
QOpenGLShader::compile(Vertex): ERROR: 0:2: '120' : version number not supported

*** Problematic Vertex shader source code ***

          #version 120
#line 2
          attribute highp vec3 vertex;
          uniform mat4 matrix;
          void main()
          {
              gl_Position = matrix * vec4(vertex, 1.0);
              gl_PointSize = 1.0;
          }

***
Echec de compilation des programmes d'ombrage OpenGL.
Essayez l'option de ligne de commande '--useGLES'ERROR: 0:2: '120' : version number not supported
QObject::connect: Cannot queue arguments of type 'QList<double>'
(Make sure 'QList<double>' is registered using qRegisterMetaType().)
Echec de compilation des programmes d'ombrage OpenGL.
Essayez l'option de ligne de commande '--useGLES'ERROR: 0:2: '120' : version number not supported
Echec de compilation des programmes d'ombrage OpenGL.
Essayez l'option de ligne de commande '--useGLES'ERROR: 0:2: '120' : version number not supported
Echec de compilation des programmes d'ombrage OpenGL.
Essayez l'option de ligne de commande '--useGLES'ERROR: 0:2: '120' : version number not supported
Echec de compilation des programmes d'ombrage OpenGL.
Essayez l'option de ligne de commande '--useGLES'ERROR: 0:2: '120' : version number not supported

```

ChristopheMineau, 2021-04-27:
Martin,
There is something weird with this last package because even with --useGLES, I have a version error:
```
PS C:\Installations\openhantek_20210426-62c54e6-1_win_x64> .\OpenHantek.exe --useGLES
Attribute Qt::AA_ShareOpenGLContexts must be set before QCoreApplication is created.
Attribute Qt::AA_UseOpenGLES must be set before QCoreApplication is created.
QOpenGLShader::compile(Vertex): ERROR: 0:2: '150' : version number not supported
ERROR: 0:2: 'in' : storage qualifier supported in GLSL ES 3.00 and above only

*** Problematic Vertex shader source code ***

          #version 150
#line 2
          in highp vec3 vertex;
          uniform mat4 matrix;
          void main()
          {
              gl_Position = matrix * vec4(vertex, 1.0);
              gl_PointSize = 1.0;
          }

***

OpenGL version "OpenGL ES 2.0 (ANGLE 2.1.0.57ea533f79a7)"
To get rid of this error message, try the command line option '--useGLSL120' or '--useGLES'
QOpenGLShader::compile(Vertex): ERROR: 0:2: '120' : version number not supported

*** Problematic Vertex shader source code ***

          #version 120
#line 2
          attribute highp vec3 vertex;
          uniform mat4 matrix;
          void main()
          {
              gl_Position = matrix * vec4(vertex, 1.0);
              gl_PointSize = 1.0;
          }

***
Echec de compilation des programmes d'ombrage OpenGL.
ERROR: 0:2: '120' : version number not supported
QObject::connect: Cannot queue arguments of type 'QList<double>'
(Make sure 'QList<double>' is registered using qRegisterMetaType().)
Echec de compilation des programmes d'ombrage OpenGL.
ERROR: 0:2: '120' : version number not supported
Echec de compilation des programmes d'ombrage OpenGL.
ERROR: 0:2: '120' : version number not supported
Echec de compilation des programmes d'ombrage OpenGL.
ERROR: 0:2: '120' : version number not supported
OpenHantek6022 has stopped after 10 s
PS C:\Installations\openhantek_20210426-62c54e6-1_win_x64>
```

Ho-Ro, 2021-04-27:
Yes, I also discovered that I forgot to commit my last change - nevertheless I added even more robustness and automatism. This will hit unstable later today.

commit 62c54e61 (2021-04-26): Improve OpenGL robustness; refactor glscope.cpp; fix #184

If the graphic system claims to support SL version 150 but this fails,
fall back automatically to legacy version 120. If this also fails,
show the hint to use the command line option '--useGLES'
to switch from OpenGL to OpenGLES (OpenGL for Embedded Systems),
which cannot be done automatically.