OpenHantek/OpenHantek6022 #185: Two-channel samplerate is not limited after start

spam-receiver, 2021-04-26:
After starting the program, it is allowed to turn on samplerate above 15MS/s until you turn off and turn on again any channel.

Ho-Ro, 2021-04-27:
I tried to reproduce this issue:
- select two channels
- select time base 1µs/div
- set sample rate e.g. 10 MS/s (or 12 or 15)
- switch off
- switch on again
- time base is 1µs/div
- sample rate is 12 MS/s
- increase sample rate 
- will not go > 15 MS/s

Please give a step-by-step instruction how to reproduce your issue - thx.

spam-receiver, 2021-04-27:
It happens only under Windows 10. In Fedora 34 Linux everything is normal.

spam-receiver, 2021-04-29:
It works with #185. Thank you.

Ho-Ro, 2021-04-29:
@spam-receiver Thx for reporting - update is available in [unstable](https://github.com/OpenHantek/OpenHantek6022/releases/tag/unstable) now.
How does the program behave the other way round? Starting with CH1 mode, can you go up to 24 and 30 MS/s?
I cannot check on a win system with real HW, only Demo Mode that starts always with 2 channels.

spam-receiver, 2021-04-29:
You right. Now one channel-mode can't go upper 15MS/s after start until on/off any channel. I forgot to check it out.

Ho-Ro, 2021-04-29:
Ok, but better this way round.

Ho-Ro, 2021-05-03:
**Tl;dr: It will take some more time...**

I created some test builds with extensive tracing output and it seems that the described behaviour is caused by a race condition between the startup default setting (always started with 1 channel in the past, now defaults to 2) and the persistent setting from the last shutdown. This sets always the correct (stored) number of channels, but misses to update the max sample rate limit according to the actual number of channels - it still uses the one from the startup value. The communication between the core and UI of the program is based on Qt's signal/slot mechanism that obviously adds some unpredictable delays.

commit 92ba5a65 (2021-05-18): fix #185

update samplerate if the limits have changed
closed 2021-05-18 by Ho-Ro