OpenHantek/OpenHantek6022 #188: Slow Main Screen Update/Refresh

stephenwang5, 2021-06-14:
The main signal screen takes ~1 sec to fresh if I change the timebase or if the signal itself changes. I encountered this when probing capacitor discharge on a 555 timer (waveform shown bellow). I would understand if this is some hardware limitation but my CPU usage appears very low and to my surprise it appears to refresh more quickly the smaller timebase I used. I've tried to adjust the minimal time between captured frames but to no avail.

**To Reproduce**
I left most of the settings on default and all the other settings are shown in the screenshot bellow.

**Expected behavior**
The screen should reflect the signal changes instantly (or to be perceived as such).

**Screenshots**
Here's the waveform I'm inspecting
[image]

**Computer environment:**
 - OpenHantek version: release 3.2.2
 - OS: Linux 5.4.0-74-generic kernel
 - Distribution, version: Ubuntu 20.04, amd64
 - Video hardware: Intel HD 530 + NVIDIA GTX 960M
 - OpenGL version: 5.12.8+dfsg-0ubuntu1 apt package
 - Qt version: 5.12.8+dfsg-0ubuntu1 apt package
 - C++ compiler version: gcc 9.3.0

**Scope device (please complete the following information):**
 - Device: 6022BE
 - Program top line OpenHantek6022 (3.2.2) - Device DSO-6022BE (FW0209)
 - Input signal: calibrated to 1kHz
 - Probe setting: same problem on both X1 and X10

**Additional context**
Program started using the `.desktop` generated when installing (executed without any additional flags)

Ho-Ro, 2021-06-14:
Stephen, the behaviour of the scope is exactly what one would expect from a DSO (digital **storage** scope). A DSO works like this: capture the input signal into a buffer (in your case at least for 200 ms/div * 10 div = 2 s) and when the buffer is filled refresh the screen with the new buffer content. This would give one refresh every two seconds for a free-running signal. But to get a standing display you want to trigger. The Hantek6022 don't have any hardware trigger, so triggering is done completely in SW. Therefore the program samples into a bigger buffer that equates twice the screen size, searches for a sufficient trigger point and displays the corresponding section of the buffer. This leads to a refresh every 4 seconds. As you wrote correctly the refresh rate increases when sampling faster.

> **Expected behavior**
The screen should reflect the signal changes instantly (or to be perceived as such).

If you want to follow **real slow** signals you should use the **Roll** trigger mode that is available for slow timebase >= 200 ms/div. Otherwise the screen refresh at e.g. 10 s/div would take 200 s!

>  the minimal time between captured frames ...

is used to reduce high CPU load - especially on small systems like RasPi - at **high** sample rates (>10MS/s).

closed 2021-06-14 by Ho-Ro

stephenwang5, 2021-06-14:
Thank you I see now!