OpenHantek/OpenHantek6022 #189: Ability to calibrate voltage reading

KanKub, 2021-06-17:
**_Is your feature request related to a problem? Please describe._**
I need corrected voltage reading. the current reading is offset by about 120mV and that's not usable when measuring some dc signal, I have to manually read the scope scale.

**_Describe the solution you'd like_**
Options to calibrate/offset the reading voltage.

**_Additional context_**
[image]
this is the signal when I connect the probe to GND.

spam-receiver, 2021-06-18:
You can do it in a text editor by changing parameters [offset] and [gain] of calibration file:

Calibration file paths:

Linux/FreeBSD/macOS:
$HOME/.config/OpenHantek/modelDSO6022.ini

Windows:
%APPDATA%\Roaming\OpenHantek\modelDSO6022.ini

This is my calibration file, if you need an example of it: [modelDSO6022.zip](https://github.com/OpenHantek/OpenHantek6022/files/6675542/modelDSO6022.zip)

Qt default settings path documentation:
https://doc.qt.io/qt-5/qsettings.html#locations-where-application-settings-are-stored

KanKub, 2021-06-18:
Thank you! but I couldn't find OpenHantek Folder in %appdata% (Windows 10). How can I find them? I also look in the installation folder at desktop but no clue of any config.

spam-receiver, 2021-06-18:
You have to create such folder and insert calibration file into it. Then you can play with [offset] and [gain] parameters by editing them. Do not forget to restart the program for the changes to be accepted.

Ho-Ro, 2021-06-18:
There's also a [calibration program](https://github.com/Ho-Ro/Hantek6022API/blob/main/examples/calibrate_6022.py) that calculates not only offset but also gain calibration values and [creates the file](https://github.com/Ho-Ro/Hantek6022API#create-calibration-values-for-openhantek) - or even better - stores the values in the device's EEPROM.

closed 2021-06-18 by Ho-Ro