OpenHantek/OpenHantek6022 #191: Provide a option to choose language?

cdhigh, 2021-06-21:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Some translations may be wrong or difficult to understand. In this case, it may be necessary to switch to English temporarily, but it is currently not possible.**

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Can provide an option in the configuration dialog to select different languages?**

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

Ho-Ro, 2021-06-22:
You already have two possibilities to change the language:
1. Use command line option `Openhantek -i` (or --international) to skip the translation completely and get the original English text.
2. Use environment variable LANG to select what you want, e.g. `LANG=fr OpenHantek` switches to French, while `LANG=C ...` disables translation at all.

The setting in the configuration dialogue cannot change the language selection immediately, a shutdown/restart is necessary to change the localisation as the texts have to be loaded when the application starts.

closed 2021-06-22 by Ho-Ro

cdhigh, 2021-06-22:
Now that there is a workaround, it can not be implemented.
Agree