OpenHantek/OpenHantek6022 #192: Return system theme

Nikita-Presnov, 2021-06-21:
I can't say that I am strongly against other topics, but fusion is really painful to look at after breeze.

closed 2021-06-22 by Ho-Ro

Ho-Ro, 2021-06-22:
The reason for Fusion is simply the smaller size on screen. Fusion (837x662) fits perfectly on the screen of an older HD1280x720 (or 1024x768) Laptop, while Breeze (837x795) collides with the statusbar. But now you can get the Breeze theme with command line option `OpenHantek -b` or`OpenHantek --breeze`, available in latest [unstable release](https://github.com/OpenHantek/OpenHantek6022/releases/tag/unstable). 

[image]

[image]

Ho-Ro, 2021-06-23:
@Nikita-Presnov No command line option anymore, you can now select "Fusion" in options Oscilloscope / Settings / Color
-> unstable

Nikita-Presnov, 2021-06-23:
Okay

commit 783c14af (2021-06-22): fix #192 - command line option -b or --breeze selects Breeze theme