OpenHantek/OpenHantek6022 #198: Issues finding macdeployqtfix.py

Yiltan, 2021-06-24:
Hello,

I am trying to install this software using the instructions from  `https://github.com/OpenHantek/OpenHantek6022/blob/v2.17-rc8/docs/build.md#macosx`

```# find all other dependencies, and their dependencies, and their... (you got it!)
python ../../macdeployqtfix/macdeployqtfix.py OpenHantek.app/Contents/MacOS/OpenHantek $(brew --prefix qt5)
```

I cannot seem to find the file `macdeployqtfix.py`.  Using the command `find $(brew --prefix qt5) -name macdeployqtfix.py` does not find this file.

From googling this is the only thing I can find about this file: https://github.com/arl/macdeployqtfix

Do I need to install this project before proceeding?

Ho-Ro, 2021-06-24:
> `https://github.com/OpenHantek/OpenHantek6022/blob/v2.17-rc8/docs/build.md#macosx`

You're using an outdated documentation (v2.17-rc8), we're already at 3.2.3.
[image]

Just follow the [latest documentation](https://github.com/OpenHantek/OpenHantek6022/blob/main/docs/build.md#macos):
````
git submodule update --init --recursive
brew update
brew install libusb fftw qt5 cmake binutils create-dmg
````
and then:
````
mkdir build
cd build
cmake ..
#
make -j4
#
# now the target was created in subdir openhantek
# .. either as single binary OpenHantek, then you're done
# .. or as a bundle if enabled in ../../openhantek/CMakeLists.txt
# .. but this bundle is still a template as the dynlibs are not yet bundled
# .. this magic will happen now
#
cd openhantek
#
# deploy all necessary Qt dynlibs into the bundle
macdeployqt OpenHantek.app -always-overwrite -verbose=2
#
# find all other dependencies, and their dependencies, and their... (you got it!)
python ../../utils/macdeployqtfix/macdeployqtfix.py OpenHantek.app/Contents/MacOS/OpenHantek $(brew --prefix qt5)
#
# finally create OpenHantek.dmg from OpenHantek.app
create-dmg --volname OpenHantek --volicon ../../openhantek/res/images/openhantek.icns --window-pos 200 120 \
  --window-size 800 400 --icon-size 100 --icon "OpenHantek.app" 200 190 --skip-jenkins \
  --hide-extension "OpenHantek.app" --app-drop-link 600 185 --eula ../../LICENSE OpenHantek.dmg OpenHantek.app
#
````

closed 2021-06-24 by Ho-Ro

Yiltan, 2021-06-24:
Thank you for the help