OpenHantek/OpenHantek6022 #207: Severe jitter causing signal corruption

kobus-v-schoor, 2021-08-18:
**Describe the bug**

I'm getting severe timing jitter on Fedora 34 (kernel 5.13.9-200.fc34.x86_64) causing the signal to be completely corrupted on higher timebases (>=1ms) - I'm trying to determine which of the following parts of my setup is causing the issue:

- I recently switched over from Debian to Fedora, I can't remember if I've successfully used the scope on Fedora before
- The scope might be faulty

I've tested adding myself to the realtime group as described in the docs, and I've also tested running OpenHantek inside an Ubuntu VM with USB passthrough (which seems to run better, although there was also some jitter, possibly caused by latencies introduced by the USB passthrough). I've also tested downgrading to a slightly older kernel version to no avail.

**To Reproduce**

Since @Vascom is the Fedora maintainer, I was wondering if they could confirm whether or not their setup is currently working as expected. It usually runs fine for about 10 seconds before becoming a problem. It seems like it might be disappearing at lower timebases (~100us) although it just might be due to the faster triggering (there are some glitches which makes me think that it still happens)

**Screenshots**

Jitter can be seen here on the second pulse right of the middle. This is the 1kHz calibration signal (which I confirmed to be working with a another scope)

[image]

**Computer environment (please complete the following information):**
 - OpenHantek version: 3.2.4
 - OS: Linux 
 - Distribution, version: Fedora Workstation 34 (using the official openhantek package, although Flatpak package had the same issue)

**Scope device (please complete the following information):**
 - Device: 6022BE
 - Program top line: OpenHantek6022 (3.2.4) - Device (FW0209)
 - Input signal: Calibration out 1 kHz
 - Probe setting: probe 1X

Vascom, 2021-08-18:
Do you use Scrolling mode?
Try switch to Normal mode.

Vascom, 2021-08-18:
In Scrolling mode I also see some artifacts like this
[image]
But in Normal mode there is no one glitch.

Ho-Ro, 2021-08-18:
Normally OH6022 captures one complete block (with double size of the visible part), searches for a trigger and displays the triggered part of the data; this takes very long for slow time base settings (1s/div = 10s/screen = 20s/capture).
Scrolling mode is intended to show these slow signals in quasi realtime. It captures small chunk after chunk with small time gaps in between, these gaps are barely visible for slow signals (e.g. 1s/div). But during this gap the USB is released and another device can grab the bus - this can delay the gap randomly and lead to jitter for faster signals.

kobus-v-schoor, 2021-08-18:
Thank you for getting back to me, to answer your questions:

- The screenshot was on normal triggering, I tried auto as well as single-shot and both produced the same jitter (you can also see from my screenshot that there was multiple points were it almost looks like packets are either lost or just somehow distorted)
- The timebase was significantly faster than 1s (1ms per/div IIRC), the screenshot was of a 1kHz signal.

I'm going to test this tomorrow with another 6022BE scope from a friend, I'll report back here if the problem was with the scope or not.

spam-receiver, 2021-08-19:
I have no such problems on Fedora 34 (kernel 5.13.9-200.fc34.x86_64). I am wondering what is the hardware configuration of your computer?

kobus-v-schoor, 2021-08-19:
I just tested another 6022BE scope on my laptop, and it had the same issue, so I'm guessing the issue lies with some combination of Fedora and my laptop's hardware (been using the scope successfully on Debian for a few months). For reference, I have an HP Probook 450 G5. I'm not seeing any errors in the dmesg logs so I don't think there is connection issues (or at least issues that the kernel is aware of) - one thing that I did notice on Fedora is that my battery lasted quite a bit longer, so I think there might be some power saving feature that is enabled that is lowering the performance of the USB bus? I tested using the "performance" mode in the power settings, still had the same issue though. I'm planning moving back to Debian anyway (Fedora is a bit too fast-paced for me), so I'll report back here if it changed anything.

Ho-Ro, 2021-08-19:
Which kind of USB do you use USB3? You could try another port or put an USB2 hub in between.

kobus-v-schoor, 2021-08-19:
Now running Debian bullseye (11) - unfortunately the problem is still here, so I have a feeling the upgraded kernel is not playing well with my hardware somehow. I unfortunately don't have a USB2 hub to test with, and all my ports are USB3 I think (I tried all of them)

Ho-Ro, 2021-08-19:
I'm also on bullseye since  (buster before), but on an old ThinkPad T400 (USB2 only). I do not see any regressions (so far).

Ho-Ro, 2021-08-19:
sorry, closed by mistake :(

kobus-v-schoor, 2021-08-19:
Ok I think I'm going to try opening a bug report with Debian to try and see if they might have some more insight into how the kernel changed w.r.t to USB (if it even is that, but I suspect that's probably the only place where this issue could originate from). Reddit said that I should maybe try using an older kernel, so I'm going to try and see if I can somehow do that as well. Could you maybe give me some insight into how the packets are handled, as this could possibly point me in the right direction with the bug report? For example:

- Is there any form of synchronization between the host and the scope, would OH realize it if there were packets lost?
- If there was latency/throughput issues on the bus and it couldn't transfer the packets fast enough, what would happen?
- Do you think there is any way that I could make use of OH to try and debug the issue (is there maybe a spot where I could print some logging to try and figure out what's going on)?

Thanks for all the help thusfar

Ho-Ro, 2021-08-20:
> - Is there any form of synchronization between the host and the scope, would OH realize it if there were packets lost?

No sync, OH communicates with the EzUSB in bulk mode. As a starter please read #60.
 
> - If there was latency/throughput issues on the bus and it couldn't transfer the packets fast enough, what would happen?

Package drop w/o notification. OH will receive the next valid packet.

> - Do you think there is any way that I could make use of OH to try and debug the issue (is there maybe a spot where I could print some logging to try and figure out what's going on)?

The communicaton takes place in `usb/scopedevice.cpp`. I've modified the verbosity level handling to capture also USB with command line option `--verbose=7` (each level adds more output, 0: quiet; 1,2: startup; 3,4: + user actions; 5,6: + data processing; 7 + USB).
The interesting part should be 

    int ScopeDevice::bulkReadMulti( unsigned char *data, unsigned length, bool captureSmallBlocks, unsigned &received, int attempts ) {

that transfers a big block by calling `bulkTransfer()` either once for a complete block (captureSmallBlocks == false -> this will be interesting for you) or in smaller chunks (captureSmallBlocks == true, set for roll mode to get faster screen response). 
`bulkTransfer()` finally calls the libusb function `libusb_bulk_transfer()` `attempts` times (where attempts is set to HANTEK_ATTEMPTS = 3).

kobus-v-schoor, 2021-08-20:
Thanks Martin, I'll set up a dev environment and see what info I can get playing around with the various parameters (thanks for adding the new verbosity level). Given that packets can be dropped w/o notification I have a feeling like it might be the issue, because what I'm seeing onscreen looks like some parts of the signal just goes missing. I'll report back here once I have something interesting (might take some time, I have a few other things I need to attend to first).

iss000, 2021-08-21:
I'm reusing this issue because my OS is Linux Fedora 34 (kernel 5.13.10-200.fc34.x86_64) and noticed recently strange behavior kind of close to the described jitter. I walked back trough last commits and found that it appears when the firmware was updated i.e. commit 3fc9afb4c104c9dd7c9b791155b444625a41db9f noted in the title bar as **commit-922**.
The problem is missing pulses when calibration freq is set to 100kHz (every lower freq is OK). The scope is 6022BL, both probes attached to calibration output.
Commit-922:
[image]

Previous commit-921 is just fine:
[image]

@Ho-Ro: Thanks for your great work!

kobus-v-schoor, 2021-08-21:
After installing an older kernel in bullseye (the newest buster kernel, 4.19.0-17), everything seems to be working again, so it seems there is a regression in the kernel. Thanks @Ho-Ro for your help, I'll use what you posted here in my bug report to the Debian team. Since the issue doesn't seem to lie with OH I'm closing for now. Thanks everyone for the help!

@iss000 Your first screenshot does indeed look like the issue I'm facing, although mine was present on v3.2.4 already (i.e. before the commit with the "updated" firmware), and mine was visible on all the lower frequencies as well. What does your hardware setup look like (i.e. your PC's details)? Maybe the same kernel regression is affecting you as well, but perhaps to a lesser extent.

Ho-Ro, 2021-08-21:
@iss000: thx, reverted to old FW build from buster with 165f31e - build 924, will be also next `unstable`

closed 2021-08-21 by Ho-Ro

kobus-v-schoor, 2021-08-21:
Found the issue after reporting the bug on the Debian BTS, details can be seen here: https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=992637

TLDR: There is a power saving feature in the kernel that was too aggressive (for my specific hardware), causing too high latency/unreliable packet transmission. Setting the "Runtime PM for PCI Device Intel Corporation Sunrise Point-LP CSME HECI #1" to "Bad" in powertop inside the "Tunables" section fixed the issue on my hardware. Thanks to everyone that helped!

Ho-Ro, 2021-08-22:
@kobus-v-schoor Thx for your activities!
@iss000 I have opened the issue above so that the topic will not be forgotten.