OpenHantek/OpenHantek6022 #219: program start fails

antonteplov, 2021-09-18:
Linux ubuntu 18.04
opengl libopengl0/bionic-updates,now 1.0.0-2ubuntu2.3 amd64 [установлен]

 sudo OpenHantek 
OpenHantek6022 (3.2.4)
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
"4.3 (Compatibility Profile) Mesa 20.0.8"
Graphic: 4.3 (Compatibility Profile) Mesa 20.0.8 - GLSL version 1.20
/usr/include/c++/7/bits/stl_vector.h:815: std::vector<_Tp, _Alloc>::const_reference std::vector<_Tp, _Alloc>::operator[](std::vector<_Tp, _Alloc>::size_type) const [with _Tp = double; _Alloc = std::allocator<double>; std::vector<_Tp, _Alloc>::const_reference = const double&; std::vector<_Tp, _Alloc>::size_type = long unsigned int]: Assertion '__builtin_expect(__n < this->size(), true)' failed.
Аварийный останов

Ho-Ro, 2021-09-18:
You shouldn't run OpenHantek as root, please do it as normal user. If you install the programm correctly either from the provided *.deb or from your own build you will have a working udev rule in place to access the scope via USB.

Ho-Ro, 2021-09-27:
9 days gone with no response from @antonteplov - either he had success or lost interest.
I will close here.

closed 2021-09-27 by Ho-Ro