OpenHantek/OpenHantek6022 #227: Program freezing

Sidddha, 2021-09-22:
Hello!
Program freezes once any signal reach the probe.

Steps to reproduce the behavior:
1. I plug in the scope
2. Touch pwm ic  or even my hand with the probe
3. Program freeze and no longer answer

**Computer environment (please complete the following information):**
 - OpenHantek version: 3.2.4
 - OS: Linux 5.11.0-34-generic
 - Distribution, version: Ubuntu 20.04.3 LTS
 - Video hardware: [AMD/ATI] Picasso (rev c1) (prog-if 00 [VGA controller])
 - OpenGL version string: 4.6 (Compatibility Profile) Mesa 21.0.3
 - Qt version: 5.12.8
 - C++ compiler version: 9.3.0

**Scope device:**
 - Device Hantek 6022BE
 - Program top line OpenHantek6022 (Build 9.2.4) - Device DSO-6022BE (FW 0209)
 - Input signal: 300kHz PWM cause failure, but 1kHz test terminals operate normal
 - Probe setting:  X1

**Additional context**
➜  ~ OpenHantek
OpenHantek6022 (3.2.4)
"4.6 (Compatibility Profile) Mesa 21.0.3"
Attribute Qt::AA_ShareOpenGLContexts must be set before QCoreApplication is created.
Attribute Qt::AA_UseOpenGLES must be set before QCoreApplication is created.
Graphic: 4.6 (Compatibility Profile) Mesa 21.0.3 - GLSL version 1.20
OpenHantek6022 has stopped after 16 s

Ho-Ro, 2021-09-22:
> Program freeze and no longer answer

No interaction possible? (e.g. change time base)
Can you resize or close the screen?
How do you terminate the program?

> Input signal: 300kHz PWM cause failure

Where and what do you measure (schematics), which voltage, etc.
What kind of computer do you use? (Laptop on bat or on charger, desktop PC?)
Did you connect the ground of the scope properly? (connect scope ground to circuit ground before starting any measurement!)

Sidddha, 2021-09-22:
Well it just trigger mode "single" was selected. I change to auto and problem gone. Seem like it call pause not freezing. 
Sorry for interrupt you

closed 2021-09-22 by Sidddha