OpenHantek/OpenHantek6022 #243: Pixel-perfect sinc interpolation

ericfont, 2021-11-19:
**Is your feature request related to a problem? Please describe.**
Not really a problem.

But if I have sinc interpolation enabled but use a Horizonotal Samplerate of >= 10 MS/s and Timebase 2 us, the sinc interpolation doesn't *look* like sinc interpolation (looks just like linear interpolation) on my big monitor:

[image]

However if I reduce Sample rate to 5 MS/s at that same Timebase of 2 us, then sinc interpolation looks like sinc interpolation:

[image]

I notice then with a few other settings....like with Timebase 5us and Samplerate 5 MS/s then sinc interpolation also doesn't *look* like sinc interpolation (seems to look just like linear):

[image]

However if I reduce Sample rate to 2 MS/s at that same Timebase of 5 us, then sinc interpolation again does look like sinc interpolation:

[image]

I don't know how the sinc interpolation is implemented, but it almost looks like it is really piece-wise linear with the horizontal distance of each line being related to the Timebase and Samplerate...  And so my somehow when the Timebase is too small and the Samplerate is too big, the sinc interpolatrion no longer looks like sinc but just looks like regular linear.

**Describe the solution you'd like**
It would be nice if the sinc-interpolation was done per horizontal pixel, rather than based on some piece-wise linear.  So on a big monitor when soomed in at low Timebase that the sinc interpolation would look like sinc interpolation.

Ho-Ro, 2021-11-20:
It is caused by [these lines](https://github.com/OpenHantek/OpenHantek6022/blob/e7f7a5d75054590ba0085a8a63bf28108ce73e21/openhantek/src/post/graphgenerator.cpp#L113), sinc is enabled if necessary (there are too less (<200) samples on the screen):
````
        // sinc interpolation if there are too less samples on screen
        // https://ccrma.stanford.edu/~jos/resample/resample.pdf
        if ( interpolationSinc && dotsOnScreen < 200 ) { // < 1 µs/div
````

ericfont, 2021-11-21:
OK, so it seems to look more like sinc when I replace that with `if ( interpolationSinc ) {`.  Seems like you made a deliberate decision to limit for less than 200 samples on screen but I don't know why...I personally would say keep using sinc interpolation when there are fewer samples.  But that is fine if it is deliberate and your decision...in that case you can close this issue.

Ho-Ro, 2021-11-21:
I think I defined the 200 pixel limit to avoid needless calculations that lead to instability, but I didn't check this in detail. Please check my hack from latest commit 71cc801.

closed 2021-11-21 by Ho-Ro

ericfont, 2021-11-21:
Just tried out that commit and seems to work fine, thanks.

commit 71cc8017 (2021-11-21): fix #243 (1st try)

do sinc interpolation only if number of samples on screen < screen width