OpenHantek/OpenHantek6022 #246: if OpenHanktek is running, and then put a Teensy into usb, then lsusb hangs and prevents me from uploading to teensy

ericfont, 2021-11-24:
**Describe the bug**
1. I have both my Hantek scope and Teensy 4.0 plugged into my computer's USB.
2. I am able to use Arduino IDE and can put my teensy into programming mode and compile & upload my teensy sketch though USB fine
3. lsusb works fine
4. But then if I open OpenHantek...I can see received scope data fine in OpenHantek, however if I press the button on my Teensy to reboot it to put it in programming mode, then the teensy just hangs with a red light.  And trying to run lsusb seems to take like 20 seconds to produce output (and when it produces output it is a bit slow).  And I can't program my Teensy via arduino IDE cause arduino IDE can't seem to detect teensy.  OpenHantek is still running fine.  But OpenHantek has somehow interferred with the peaceful behavior of my computers connection to the other usb devices in a way I can't explain.  I'm noticing now that even if I don't have my Teensy plugged in and start OpenHantek, lsusb seems to work fine.  But when I plug in teensy after starting openhantek, then lsusb is back to having this problem of taking 20 seconds to complete.

I'm on Ubuntu 20.04 LTS (KDE Neon).  I will have to investigate some more tomorrow.  But it seems the presence of both OpenHantek and teensy seems to cause lsusb to take a long time and causes teensy to not reboot properly into programming mode.

Ho-Ro, 2021-11-24:
The Hantek HW can saturate the USB when running, but I never encountered a behaviour like you. Please provide the output of ``lsusb; lsusb -t`` to see if both devices share the same bus.

ericfont, 2021-11-24:
Aha problem goes away if I plug in the Hantek into a different port that is at it's own root of the tree.  So for instance this configuration works fine:

```
e@KDE-neon-Ubuntu-LTS:~$ lsusb -t
/:  Bus 04.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 5000M
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 480M
    |__ Port 4: Dev 3, If 0, Class=Vendor Specific Class, Driver=usbfs, 480M
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 10000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/10p, 480M
    |__ Port 5: Dev 2, If 0, Class=Hub, Driver=hub/4p, 480M
        |__ Port 1: Dev 9, If 0, Class=Communications, Driver=cdc_acm, 480M
        |__ Port 1: Dev 9, If 1, Class=CDC Data, Driver=cdc_acm, 480M
        |__ Port 4: Dev 5, If 0, Class=Human Interface Device, Driver=usbhid, 12M
        |__ Port 4: Dev 5, If 1, Class=Human Interface Device, Driver=usbhid, 12M
    |__ Port 6: Dev 3, If 0, Class=Hub, Driver=hub/7p, 480M
        |__ Port 3: Dev 4, If 0, Class=Human Interface Device, Driver=usbhid, 1.5M
        |__ Port 3: Dev 4, If 1, Class=Human Interface Device, Driver=usbhid, 1.5M
```

(where `Port 4: Dev 3, If 0, Class=Vendor Specific Class, Driver=usbfs, 480M` is the Hantek and `Port 1: Dev 9, If 1, Class=CDC Data, Driver=cdc_acm, 480M` is the teensy).

But the following configuration I had earlier doesn't work:

```
e@KDE-neon-Ubuntu-LTS:~$ lsusb -t
/:  Bus 04.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 5000M
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 480M
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 10000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/10p, 480M
    |__ Port 5: Dev 2, If 0, Class=Hub, Driver=hub/4p, 480M
        |__ Port 1: Dev 24, If 0, Class=Human Interface Device, Driver=usbhid, 480M
        |__ Port 4: Dev 5, If 0, Class=Human Interface Device, Driver=usbhid, 12M
        |__ Port 4: Dev 5, If 1, Class=Human Interface Device, Driver=usbhid, 12M
    |__ Port 6: Dev 3, If 0, Class=Hub, Driver=hub/7p, 480M
        |__ Port 5: Dev 26, If 0, Class=Vendor Specific Class, Driver=usbfs, 480M
        |__ Port 3: Dev 12, If 0, Class=Human Interface Device, Driver=usbhid, 1.5M
        |__ Port 3: Dev 12, If 1, Class=Human Interface Device, Driver=usbhid, 1.5M
```

And when I run that with the teensy in program mode, the output takes like 20 seconds and looks like:

```
e@KDE-neon-Ubuntu-LTS:~$ lsusb -t
/sys/bus/usb/devices/1-5.1:1.1/bAlternateSetting: No such file or directory
/sys/bus/usb/devices/1-5.1:1.1/bInterfaceClass: No such file or directory
/sys/bus/usb/devices/1-5.1:1.1/bInterfaceNumber: No such file or directory
/sys/bus/usb/devices/1-5.1:1.1/bInterfaceProtocol: No such file or directory
/sys/bus/usb/devices/1-5.1:1.1/bInterfaceSubClass: No such file or directory
/sys/bus/usb/devices/1-5.1:1.1/bNumEndpoints: No such file or directory
/:  Bus 04.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 5000M
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 480M
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 10000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/10p, 480M
    |__ Port 5: Dev 2, If 0, Class=Hub, Driver=hub/4p, 480M
        |__ Port 1: Dev 21, If 0, Class=Human Interface Device, Driver=usbhid, 480M
        |__ Port 1: Dev 21, If 1, Class=, Driver=, 480M
        |__ Port 4: Dev 5, If 0, Class=Human Interface Device, Driver=usbhid, 12M
        |__ Port 4: Dev 5, If 1, Class=Human Interface Device, Driver=usbhid, 12M
    |__ Port 6: Dev 3, If 0, Class=Hub, Driver=hub/7p, 480M
        |__ Port 5: Dev 22, If 0, Class=Vendor Specific Class, Driver=usbfs, 480M
        |__ Port 3: Dev 12, If 0, Class=Human Interface Device, Driver=usbhid, 1.5M
        |__ Port 3: Dev 12, If 1, Class=Human Interface Device, Driver=usbhid, 1.5M
```

closed 2021-11-24 by ericfont

ericfont, 2021-11-24:
Now that the teensy and Hantek are comming from different root buses, seems to work ok:

```
e@KDE-neon-Ubuntu-LTS:~$ lsusb -t
/:  Bus 04.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 5000M
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 480M
    |__ Port 2: Dev 4, If 0, Class=Hub, Driver=hub/4p, 480M
        |__ Port 1: Dev 7, If 0, Class=Human Interface Device, Driver=usbhid, 480M
        |__ Port 4: Dev 6, If 1, Class=Human Interface Device, Driver=usbhid, 12M
        |__ Port 4: Dev 6, If 0, Class=Human Interface Device, Driver=usbhid, 12M
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 10000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/10p, 480M
    |__ Port 6: Dev 3, If 0, Class=Hub, Driver=hub/7p, 480M
        |__ Port 3: Dev 12, If 0, Class=Human Interface Device, Driver=usbhid, 1.5M
        |__ Port 3: Dev 12, If 1, Class=Human Interface Device, Driver=usbhid, 1.5M
        |__ Port 5: Dev 26, If 0, Class=Vendor Specific Class, Driver=usbfs, 480M
```

So it doesn't seem to matter even that they are bot in non-root hubs, but as long as they originate from different root hubs they seem to functiton together.

Ho-Ro, 2021-11-24:
The user manual states this on page 2:

> If the upload takes more than 30 s, please
close the window and restart the program.
This can happen in rare cases when another high
speed USB device blocks the same USB bus.
If this happens, use another USB port of the
computer or disconnect the other device.
The scope should be connected alone to a HS-
USB bus in order to work undisturbed.
Under Linux you can check this with:
lsusb; lsusb -t

I will add to the manual that the Hantek can also block other units on the same bus.