OpenHantek/OpenHantek6022 #256: Empty window on Pine Phone

sagb, 2022-01-04:
I try to run OpenHantek on Pine Phone (arm64).
It starts, some grid lines flash briefly (seems like borders of widgets, no content), then it ends up showing black window.

Tried:
- window managers: dwm, ratpoison (I long used OpenHantek in ratpoison on laptops without problems)
- OpenHantek 3.1.1
- `--demoMode`, `--useGLES`, `--useGLSL120`, specific `--font`

**Computer environment:**
 - OpenHantek version: 20220101 - commit 944 (3.2.5-6-g27eb8c5). Also tried 3.1.1.
 - OS: Linux
 - Distribution, version: Debian bookworm (testing)
 - Video hardware: Mali400
 - OpenGL version: 2.1 Mesa 21.2.6
 - Qt version: Qt 5.15.2
 - C++ compiler version: gcc 11.2.0

**Scope device:**
 - Device: Hantek DSO 6022BL (`--demoMode` doesn't work either)
 - Program top lines: OpenHantek6022 (20220101 - commit 944), use device "DSO-6022BL" serial number "CE24011FCD29"
 - Input signal: N/A
 - Probe setting: N/A

**Additional context**

Beginning of output from `OpenHantek --resetSettings -i --verbose 15`: https://pastebin.com/EaZ1FPMY

sagb, 2022-01-05:
Tried separate X11 client and server:

- run OpenHantek on Pine phone, displaying to remote X server
- run OpenHantek on laptop, displaying on Pine X server

Surprisingly, both work.
Combination "both executable and X server on Pine phone" still fails.

sagb, 2022-01-05:
Sometimes random mouse click on black window reveals a widget:
[image]
[image]

Ho-Ro, 2022-01-05:
Please play around with the OpenGL version as [I wrote some time ago](https://github.com/OpenHantek/openhantek/issues/221#issuecomment-462728423). Then you could try also the option `--useGLSL150`.

sagb, 2022-01-07:
> MESA_GL_VERSION_OVERRIDE=3.2 MESA_GLSL_VERSION_OVERRIDE=150 OpenHantek -i --useGLSL150
> MESA_GL_VERSION_OVERRIDE=3.2 MESA_GLSL_VERSION_OVERRIDE=150 OpenHantek -i
> MESA_GL_VERSION_OVERRIDE=4.5 OpenHantek -i
> MESA_GL_VERSION_OVERRIDE=1.0 OpenHantek -i

All failed.

sagb, 2022-01-07:
Found a workaround (but slow - unusable): forwarding X11 calls via ssh:

> ssh -X localhost OpenHantek

Ho-Ro, 2022-03-14:
No more activities, I will close. We can reopen if someone is willing to support.