OpenHantek/OpenHantek6022 #257: Background of marker selected rectangle area is transparent on Wayland

ericfont, 2022-01-10:
[image]

OpenHantek commit 944
I'm on Arch Linux 5.10.90-1-lts with latest packages from repos (wayland v1.20.0-1, xorg-xwayland v21.1.4-1, KDE plasma-desktop 5.23.5-1).

Maybe should report this to xorg-wayland people instead of you, cause it should be their repsponsiblity.  (and maybe this also has to do with opengl too).

ericfont, 2022-01-12:
I just noticed this issue on another program that use opengl, I think...so I'm sure this issue has to do with wayland and not your problem:

[image]

closed 2022-01-12 by ericfont

ericfont, 2022-01-12:
For anyone reading, I can force usage of Xorg backend for Qt (so can have the opengl working) in a Qt app under a wayland session by having the environment variable `QT_QPA_PLATFORM=xcb` set when runnning...

Ho-Ro, 2022-01-12:
Thx für investigating.